FreeCAD/FreeCAD #10894: [Problem] TechDraw Cosmetic Thread Lines

4023086, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Tech Draw cosmetic lines for internal threads are spaced around a hole by constant factor of 1.176. This factor is hard coded in the source code - see below.
In Part Design and Part, I draw holes for threads reduced by this factor, so that in Tech Draw 'add cosmetic thread hole bottom view' and 'side view' give nominal diameters -  e.g. for an M6-hole the diameter in Part Design sketch will be 6/1.176 mm. Sometimes cosmetic thread diameters in Tech Draw are slightly inconsistent. E.g. M6-hole, drawn as 6/1.176=5.102 mm : cosmetic thread hole bottom view gives a diameter of 6.01 mm, while thread hole side view shows the expected value of 6 mm (5.102*1.176= 5.999952). This inconsistency is porbably caused by different factors in the source code as shown below:

File src/Mod/TechDraw/Gui/CommandExtensionPack.cpp

line 368       _createThreadLines(SubNames, objFeat, 1.176f);

line 472       _createThreadCircle(Name, objFeat, 1.177f);

Unless there is a good reason to keep different factors, I suggest to use the same factor for both functions.

Workaround: use Format Spec  %.0w

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods:
  * freecad.gears 1.0.0
  * QuickMeasure 2022.10.28
  * sheetmetal 0.3.0
  * Assembly4 0.50.2
  * parts_library
  * symbols_library
  * fasteners 0.4.66
  * A2plus 0.4.60n
  * ThreadProfile 1.87.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[Screenshot_2023-10-02_15-00-05.png.zip](https://github.com/FreeCAD/FreeCAD/files/12782798/Screenshot_2023-10-02_15-00-05.png.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-11:
Fixed up original post by adding the image directly and also adding direct links in to the source code for convenience.

pierreporte, 2025-04-29:
A **150 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.