FreeCAD/FreeCAD #10896: [Problem] Materials Core Development

davesrocketshop, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Core features development

- Internal APIs
- diff feeature
- export inherited models in conjunction with the Materials Editor
- unit tests

### Full version info

```shell
any
```

### Subproject(s) affected?

None

### Anything else?

Part of https://github.com/FreeCAD/FreeCAD/issues/10174

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-10-04:
It looks like Material is currently implemented as a workbench even though it doesn't provide any toolbars or menus.
I suspect that this would be better implemented as a module.

The primary interface is going to be the material editor which will be accessible from many places in FreeCAD.  Let's discuss.  If this needs to be changed to a module, I'll add a new card/issue for it.

davesrocketshop, 2023-10-04:
Honestly I hadn't considered a module, I've been in the workbench mindset for so long. But yes, I think you're right, certainly as it is there anyways.

It could always me made a workbench later if there's enough functionality to warrant it. I've added a button to bring up the editor in my dev branch but it's looking pretty lonely. This is useful, but is it enough for a workbench?

yorikvanhavre, 2023-10-04:
TBH the only difference is that there is no workbench defined. All workbenches are modules, not all modules define a workbench. If you remove the workbench definition, you have a module without a workbench.

I would keep the workbench for now, because it will be easier to test features and make people aware of the new functionality.

sliptonic, 2023-10-04:
I would like to know the answer to this too.  The Ondsel Addon is not a workbench.  We currently don't need preferences but there are several features I want to add that would necessitate preferences.

WandererFan, 2023-10-05:
You could have a trivial Mod/Material/Gui/MaterialGui.cpp that only  registers your preference page.  I don't know the python for it off the top of my head.

davesrocketshop, 2023-10-05:
That was there. It didn't show until MaterialGui was loaded externally.

WandererFan, 2023-10-05:
App/Application and Gui/Application both load modules at start up.

yorikvanhavre, 2023-10-09:
There is no need to have a workbench to add a preferences page. See for ex. https://github.com/FreeCAD/FreeCAD-Help (you can just add it into InitGui.py)

maxwxyz, 2024-06-28:
could this issue be closed as implemented or is there something left?

maxwxyz, 2024-07-01:
@davesrocketshop could this issue be closed?

davesrocketshop, 2024-07-10:
Yes. Closing.

closed 2024-07-10 by davesrocketshop