FreeCAD/FreeCAD #10898: [Problem] Material Libraries API

davesrocketshop, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Create an API for interfacing with external libraries of materials and material properties. Example libraries could be based on web services/databases/git etc.

This API is intended for add on workbenches

### Full version info

```shell
Any
```

### Subproject(s) affected?

None

### Anything else?

Part of https://github.com/FreeCAD/FreeCAD/issues/10174

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-10-03:
I'm moving this to post 1.0.   If someone thinks it's a show-stopper we can pull it in.

davesrocketshop, 2025-02-17:
Still active

maxwxyz, 2025-03-09:
@davesrocketshop is this closed by https://github.com/FreeCAD/FreeCAD/pull/19908 ?

davesrocketshop, 2025-03-09:
> [@davesrocketshop](https://github.com/davesrocketshop) is this closed by [#19908](https://github.com/FreeCAD/FreeCAD/pull/19908) ?

That's only part1. There are more parts that will need to be merged before this is complete. But yes, it is part of this.

PR #21047 (2025-04-30): Materials: External Module Support
The final PR for the external module feature that allows materials to be stored in an external datastore, webservice, etc.

This includes the final material manager classes, and the UI support in the form of commands and preference pages.

## Issues
Closes #10898

## Before and After Images

The only UI change is the addition of a preference page that allows the user to enable/disable external modules, as well as view some basic caching stats.

[image]
closed 2025-04-30 by chennes