FreeCAD/FreeCAD #10900: [Problem] Materials Editor

davesrocketshop, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

User interface to edit/save/load material cards

### Full version info

```shell
Any
```

### Subproject(s) affected?

None

### Anything else?

Part of https://github.com/FreeCAD/FreeCAD/issues/10174

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2023-10-22:
New PR #11138. Primarily editor enhancements to support arrays, but also improvements to the material save workflow. Numerous minor improvements

maxwxyz, 2024-06-28:
@davesrocketshop  think this can be closed as there is a material editor.
There are also a lot of other material issues from autumn 20203, which of them could be closed as well?

closed 2024-06-28 by maxwxyz