FreeCAD/FreeCAD #10901: [Problem] Materials diff viewer

davesrocketshop, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A visual tool to compare material cards. This includes both structure (properties) and data.

It may be desirable to integrate this with the material editor and/or as a standalone. The latter should be callable from appropriate places within the application, and from the Materials workbench UI

### Full version info

```shell
Any
```

### Subproject(s) affected?

None

### Anything else?

Part of https://github.com/FreeCAD/FreeCAD/issues/10174

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-10-03:
I assume that we want this to go beyond merely comparing two cards.   Ideally this would be useful to merge attributes from multiple cards into one.

Since the material data is stored in Yaml, I assume a tool like Meld would be a good analogy.

Could this be an addon?  Could it use an external tool like meld to do the comparison.

sliptonic, 2023-10-03:
I'm marking this as post 1.0 since a lack of a diff viewer wouldn't hold up a release.  Though it would certainly be desirable to have one in place.

sliptonic, 2023-10-03:
FYI there's a macro for merging materials.  This might be useful for a starting point or should be revisited and the wiki revised if not.
https://wiki.freecad.org/Macro_merge_duplicate_materials