FreeCAD/FreeCAD #10902: [Problem] Embedded materials

davesrocketshop, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Material cards are normally saved in files. It is desirable to be able to save them and their properties into a FreeCAD document so that they can be more easily exchanged.

Some other useful functions:
- export embedded materials to the local system
- Create in-document materials that have no external file

### Full version info

```shell
Any
```

### Subproject(s) affected?

None

### Anything else?

Part of https://github.com/FreeCAD/FreeCAD/issues/10174

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-10-03:
We already have a Material object: https://github.com/FreeCAD/FreeCAD/blob/master/src/App/MaterialObject.h
It uses an App::PropertyMap to store string:string pairs. Maybe we need another property type here, that can also hold categories?

davesrocketshop, 2023-10-03:
While that object may be a suitable basis for development it is not sufficient as is. For example, it has no reference to the original material card. It has now way to determine if it contains a complete model (e.g. linear elastic properties all populated) which is possible with the extended API, and so on. Arrays may not be appropriate for property maps.

It also has none of the property model or inheritance information, so that would all need to be added.

My first impression though is that much of this can be added while keeping the current functionality.

yorikvanhavre, 2023-10-03:
Yes, I would say all this info can be added just by adding the needed properties to it.
Bigger problem IMHO is the property categories. How to represent that...

Now another idea germinates into my mind. What if each material property was also a property of the material object (dynamically added/removed when needed). You could edit it right form the task panel.

The question is, are the current object properties enough to represent all material properties...

davesrocketshop, 2025-01-27:
Still pending