FreeCAD/FreeCAD #10904: [Problem] Appearence materials

davesrocketshop, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Integrate appearance with the updated materials system

- update Part:Material property
 - Use new material cards for appearance property descriptions
 - Remove hard coded appearance values
- add texture/image support. Currently this can be done from "View"->"Texture Mapping..." but the feature doesn't work correctly and is not integrated

### Full version info

```shell
Any
```

### Subproject(s) affected?

None

### Anything else?

Part of https://github.com/FreeCAD/FreeCAD/issues/10174

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-10-04:
> * add texture/image support. Currently this can be done from "View"->"Texture Mapping..." but the feature doesn't work correctly and is not integrated

That would be awesome. Proper support for textures is much needed in FreeCAD. Hopefully, more advanced rendering will let us make full use of the appearance properties in the future.

Do you plan to add default appearances reflecting the type of material like in commercial CAD software ?

[image]

[image]

davesrocketshop, 2023-10-04:
That would be the goal. I'll need a lot of help to get there though

luzpaz, 2023-10-06:
Maybe ropping in @howetuft from Render WB in on this discussion ?

howetuft, 2023-10-06:
Thank you @luzpaz . Indeed, I've already implemented shader management, including textures, for Render WB and I've based it on FC materials. It would be convenient to converge, IMHO.

Please note also that one shouldn't implement textures standalone as, in physical based rendering (PBR), textures do not exist without shaders. Also note that efficient texture implementation for PBR should allow to apply several textures to the same shader, to its various channels (color, metal, normal, ambient occlusion... see for instance https://www.cgbookcase.com/).

maxwxyz, 2023-10-21:
please consider https://github.com/FreeCAD/FreeCAD/issues/11113
All glossy reflecting materials currently appear white when you view the face from their normal direction. This is annoying especially in sketches when you cannot differentiate different colors or materials as all appear white. Maybe the default light position/direction can be changed or even be set in the preferences

maxwxyz, 2024-07-01:
@davesrocketshop could this issue be closed?

davesrocketshop, 2024-07-10:
Yes. Closing

closed 2024-07-10 by davesrocketshop