FreeCAD/FreeCAD #10905: [Problem] Material Appearance Preview

davesrocketshop, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Materials Editor has a feature that allows the user to preview the appearance by applying the appearance properties, or at least it should. The current functionality is copied from the previous editor and uses only the diffuse color and shininess values.

Ideally, this viewer should create an OCC object similar to what would be shown in a document and apply all supported properties, including textures.

### Full version info

```shell
Any
```

### Subproject(s) affected?

None

### Anything else?

Part of https://github.com/FreeCAD/FreeCAD/issues/10174

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-10-03:
I would think we need to embed visual properties into our material cards, and modify the rendering system of FreeCAD to take these values from there instead of the viewprovider's Material property...

yorikvanhavre, 2023-10-03:
Then I guess we're on the right track 😄

davesrocketshop, 2024-03-20:
[image]

I never marked this at the time, but this feature has been implemented.

closed 2024-03-20 by davesrocketshop