FreeCAD/FreeCAD #10906: [Problem] Materials Arch Workbench migration

davesrocketshop, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Update the Arch workbench to use the new Materials system

### Full version info

```shell
Any
```

### Subproject(s) affected?

None

### Anything else?

Part of https://github.com/FreeCAD/FreeCAD/issues/10174

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-10-03:
The fix in arch itself is really easy, we should change https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Arch/ArchMaterial.py#L716 to this:
```python
import MatGui
FreeCADGui.runCommand("Materials_Edit")
```
but of course it makes not much sense at the moment, as we cannot attribute a material to an object yet

davesrocketshop, 2023-10-03:
Also, for the first PR I wanted to make the transition seamless. This entry point was only there for compatibility and is fully expected to go away

maxwxyz, 2024-07-01:
@davesrocketshop could this issue be closed?

Roy-043, 2024-07-08:
What is planned here exactly? Will Arch materials and multi materials be replaced? Or will they be kept? I am asking because I have a PR prepared to fix the colors of Arch Windows. But that code relies on the 'old' materials.

yorikvanhavre, 2024-07-09:
Materials in Arch/BIM are basically a very simple "property bag" object that is linked to a BIM object. I believe that is good for now (I mean, for 1.0)

Now, after that, I guess we have two possibilities from here:

1) We discard completely that system, and BIM objects begin to rely on pure materials from the Materials module.

2) We keep the current "property bag" object, and simply that object becomes the "holder" for a material from the Materials module

Of course 1) is probably cleaner, but we need to define better what we need from materials to make sure we can do it with Materials materials. To begin a list, I would say:

* Colors
* A hatch pattern
* One or more classification system references (That is, a reference system ex: UniClass + an ID ex: 105367)
* Custom properties (such as IFC Step ID, but possibly other IFC custom properties)
* Be contained in the FreeCAD file and editable

When working with NativeIFC, the idea is that there is an IFC file attached to the FCStd document, and everything that happens in the FCStd document is driven from the IFC file. Materials too. So here the idea would be to "reveal", or "create on the fly" materials from the contents of the IFC file when we load it.

I would say from my point of view that's more or less it. The multimaterials system in Arch/BIM is not really a materials system, and I guess it can stay as it is now.

davesrocketshop, 2024-07-10:
@yorikvanhavre has the final say here. This was to ensure that Arch/BIM was considered and it was. Closing.

closed 2024-07-10 by davesrocketshop

Roy-043, 2024-07-11:
IMO this issue has been closed prematurely. There are still several color problems related to the new material system that have to be addressed.

yorikvanhavre, 2024-07-11:
> There are still several color problems related to the new material system that have to be addressed.

But is this issue the right place for it? Wouldn't it be better to open specific issues?

davesrocketshop, 2024-07-11:
I agree with @yorikvanhavre on this one. The issues are not Arch/BIM issues.