FreeCAD/FreeCAD #10910: [Problem] Authoring and Licence, Empty Licence URL breaks

kroko, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If _Default license_ is set to `Other` and  _License URL_ is left empty, it is impossible

to create new document, throws

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: array::at
```

or open existing one, throws

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: Invalid project file /Users/path/to/doc.FCStd: array::at
```

Expected - Probably should ignore empty field.

[image]

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-10-02:
Confirmed.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34387 (Git)
Build type: Release
Branch: master
Hash: d95b26dfb5dd23eaee004b4af3cf5cf4766e7f9c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

pitch-circle, 2023-10-05:
Mentioned on the forum here:

https://forum.freecad.org/viewtopic.php?p=703544

PR #10947 (2023-10-09): Fix #10910 Authoring and Licence, Empty Licence URL breaks
Fix #10910
commit 191aee56 (2023-10-09): Fix #10910
commit 198e7ef9 (2023-11-11): Fix #10910
closed 2023-10-09 by wwmayer