FreeCAD/FreeCAD #10911: [Problem] Included VS Code setting are intrusive

kadet1090, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/pull/9098 introduced common settings for Visual Studio Code editor - it is aimed to simplify build systems for newcomers. Although it probably does it job well it is very intrusive, as it forces users to use one particular environment and settings, which cannot be overridden in any way. Some plugins like "QT Tools" also writes to the `.vscode` directory, especially `launch.json` file, so I had to force git to ignore changes. 

https://github.com/FreeCAD/FreeCAD/pull/9663 will improve situation a little bit, but I still will have to ignore changes made to files (I prefer to use Ninja with clang + clangd + lldb which gives me better experience, also QT Tools writes to that file).

I propose to include only essential settings that should be enforced in the `.vscode` folder (for me it should be only `extensions.json`) and rest should be placed inside `contrib/vscode` directory to be manually (or via some installation shell script) copied by users.

### Full version info

```shell
irrelevant
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

https://stackoverflow.com/questions/32964920/should-i-commit-the-vscode-folder-to-source-control

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-10-02:
Comments, @Pesc0?

Pesc0, 2023-10-02:
I get it, I also don't like when things are "enforced" and not the way I want, and the fact that conda is currently enforced is in fact the very reason why I started working on that PR. However:
> I propose to include only essential settings that should be enforced in the .vscode folder (for me it should be only extensions.json)

is too radical I think. What I'm trying to do with that PR is to simplify things as much as possible for beginners, basically providing good defaults, and at the same time allowing both for conda and "normal" dependency management.  I was also writing a guide [here](https://github.com/FreeCAD/DevelopersHandbook/pull/52) (however that is now outdated). I have spent a lot of time getting the debugger to work, and now with the provided `launch.json` in one click you can debug cpp and python simultaneously. I even upgraded the python debugger in FC. This would basically blast all of that out.

Also 
>  should be placed inside contrib/vscode directory to be manually (or via some installation shell script) 

is not great IMO. it's another extra script, buried in another extra directory, that users won't find, then you have to inform users that it exists, then instruct them on what to do, then someone who just wanted to work on something now has to deal with the sidequest of setting up their IDE, and then everybody gives up. ok maybe that's an exaggeration, but you get what I mean. 

Point is that this config should be provide sane defaults / a good starting point and just get out the way.
I think it's a great idea to add `.vscode` to `.gitignore` so that everyone starts out with the default folder, but then is free to customize their setup independently without having any problems with git. I think this would satisfy your needs for a different debug config as well. Could that work?

---

Then there's @oursland argument for sticking with just conda. You can read more in my PR's thread, but the gist of it is that by providing just a single "enforced" configuration you provide a "standard" way of doing things, in this case managing dependencies with conda, thus reducing the variables, and ultimately having to provide less support to users who are having problems, because now the supported config is only one and not 1000.

oursland, 2023-10-02:
> Point is that this config should be provide sane defaults / a good starting point and just get out the way.

I agree with this.

> I think it's a great idea to add .vscode to .gitignore so that everyone starts out with the default folder, but then is free to customize their setup independently without having any problems with git.

We can do this, and ensure that only intentional changes to the VSCode configuration / workspace get committed.  I believe this addresses @kadet1090 concerns with plugins like "QT Tools" altering the `.vscode` files.

> You can read more in my PR's thread, but the gist of it is that by providing just a single "enforced" configuration you provide a "standard" way of doing things, in this case managing dependencies [...]

I believe this is the easiest path forward, until environment configuration is fixed in the upstream `vscode-cmake-tools` to support environment setting per CMake Kit as noted in this issue: [#2301](https://github.com/microsoft/vscode-cmake-tools/issues/2301)

kadet1090, 2023-10-03:
>Then there's @oursland argument for sticking with just conda. You can read more in my PR's thread, but the gist of it is that by providing just a single "enforced" configuration you provide a "standard" way of doing things

In general I'm in favor of this approach, but it needs two things:
1. it has to work well - which for me is not the case, as I was unable to get conda setup to work properly, and conda is REALLY slow for some reason in my setup - trying to install everything took like 40 minutes. Which is even worse it still did not work properly (was not able to find MPI) where my system setup just works.
2. it must have escape hatch and be configurable

Current solution lacks the second point and for me even first - but I'm not saying that this is the case for everyone.

>I think it's a great idea to add .vscode to .gitignore so that everyone starts out with the default folder, but then is free to customize their setup independently without having any problems with git.

That indeed'd solve the issue. However git does not work that way, you can't ignore file in the repository without magic commands like `git update-index --assume-unchanged` and they are from my experience prone to errors. The standard way that I know is to create template files (often `.dist`) that can be copied after clone or if app supports then load `.local` files with higher precedence. https://stackoverflow.com/questions/10755655/git-ignore-tracked-files

kadet1090, 2023-11-26:
Yep, confirmed.

maxwxyz, 2024-03-02:
This seems to be resolved.

closed 2024-03-02 by maxwxyz