FreeCAD/FreeCAD #10916: [Problem] TechDraw WeldSymbol icon is unintuitive

FEA-eng, 2023-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The current TechDraw WeldSymbol icon represents a welding helmet but it's rather confusing for new users. There was a discussion about changing it to something more intuitive some time ago and @WandererFan agreed to change the icon but the OP didn't provide a proper svg file. I'd like to bring this topic back. I think that the icon should be similar to what the OP suggested but without the small details. So either a fillet weld symbol similar to this:

[image]

or a weld annotation symbol like this:

[image]

Not both at the same time to avoid small details.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34506 (Git)
Build type: Release
Branch: master
Hash: 3ce68745dc3aa3b02de0df2d5e4409bf3e286c49
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-10-03:
Because of this, I wanted to check the FreeCAD icon palette mentioned in the wiki but there seems to be an issue with access: https://forum.freecad.org/viewtopic.php?t=81649

WandererFan, 2023-10-03:
I can't fix the permissions, but I have a copy.  Remove the junk at the end of the file name.

[FreeCAD-Tango.gpl.not.txt](https://github.com/FreeCAD/FreeCAD/files/12792502/FreeCAD-Tango.gpl.not.txt)

WandererFan, 2023-10-03:
Found this in my files, too.

[image]

FEA-eng, 2023-10-03:
@WandererFan I'm definitely not an artist so I just modified the icon for the surface finish symbols tool:

[image]

Is it ok ? It should fulfill the requirements since it's based on an existing icon.

WandererFan, 2023-10-03:
This is what it looks like in a toolbar at large (32px) and medium (24px) sizes.  Not sure I'd recognize what it is at the smaller size, but I don't make drawings for welders.

[image]
[image]

WandererFan, 2023-10-03:
I'm no artist either as you can see: 

[image]

FEA-eng, 2023-10-04:
> This is what it looks like in a toolbar at large (32px) and medium (24px) sizes. Not sure I'd recognize what it is at the smaller size, but I don't make drawings for welders.

It should look better with smaller sizes now:

[image]

I think that it's pretty recognizable for those who need it (and thus easily recognize the typical weld annotation symbol) but here's an alternative:

[image]

I think that it should show a simplified representation and hence the triangle (your icon is nice but the arc can be confusing since it's not a typical symbol of a fillet weld)

Still, I prefer the first variant. It's similar to surface finish symbol and thus easier to find (after all, those annotations are quite similar in drawings).

WandererFan, 2023-10-04:
Here is your latest in a toolbar at medium:
[image]

AFAIK, the fillet symbol is supposed to be hollow.  This is yours with hollow fillet and with margins and lines thicknesses adjusted.  I like it better, but I'm not passionate about it.  
[image]

I'll update with whichever one you prefer.

FEA-eng, 2023-10-04:
Strange, the arrow disappears for some reason. Do you know what can be the cause ? It was created as a type of line end marker in Inkscape.

Can you try this one ? I drew the arrow manually here.

[image]

> AFAIK, the fillet symbol is supposed to be hollow.

I know but I just thought that it looked better as filled ;-) However, I will keep it hollow now.

WandererFan, 2023-10-04:
[image]

I'm going to go with this one.

WandererFan, 2023-10-05:
Completed by PR #10936

closed 2023-10-05 by WandererFan

luzpaz, 2023-10-06:
Anyone volunteering to update the docs ?

FEA-eng, 2023-10-06:
> Anyone volunteering to update the docs ?

Wiki is already updated: https://wiki.freecad.org/File:TechDraw_WeldSymbol.svg