FreeCAD/FreeCAD #10917: [Problem] AssertionError during build attempt

InRiPa, 2023-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Attempting to build FreeCAD under Arch Linux, using the [AUR package](https://aur.archlinux.org/packages/freecad-git#comment-936643)

results in an error:
```bash
======================================================================
FAIL: testCalculiXSteel (TestMaterialsApp.MaterialTestCases.testCalculiXSteel)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/user/.cache/yay/freecad-git/src/FreeCAD/build/Mod/Material/TestMaterialsApp.py", line 166, in testCalculiXSteel
    self.assertEqual(properties["Density"], "7900.00 kg/m^3")
AssertionError: '7900,00 kg/m^3' != '7900.00 kg/m^3'
- 7900,00 kg/m^3
?     ^
+ 7900.00 kg/m^3
?     ^

----------------------------------------------------------------------
Ran 1184 tests in 55.808s

FAILED (failures=1, skipped=2)
==> ERROR: A failure occurred in check().
    Aborting...
 -> error making: freecad-git-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
freecad-git - exit status 4
```

Maybe someone can't point me in a direction on how to solve this.

### Full version info

```shell
No Version information since I can't get it to build.
```

### Subproject(s) affected?

None

### Anything else?

Since it seems to be an issue with number representation, please let me know if you need any system details.

```bash
# cat /etc/locale.conf
LANG=en_US.UTF-8
```
My system runs in `English` / `en_US`, but for Formats I'm using `German` / `de_DE`

I'm using Arch Linux with Linux 6.5.5 and Gnome 44.5

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-10-03:
@InRiPa, could you provide `locale` output? If you set everything to `en_US.UTF-8`, does it compile?

I think you need to set only `LC_NUMERIC` to `en_US.UTF-8`.

Syres916, 2023-10-03:
I would suggest the Material test should be fixed as per https://github.com/FreeCAD/FreeCAD/compare/master...Syres916:FreeCAD:Materials_Test_Locale rather than the user having to change locale in order to compile. Not having a German test machine, I've done my best using reverse logic for testing.

InRiPa, 2023-10-03:
@xtemp09 you're right.

my `locale` before:
```
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=de_DE.UTF-8
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```
Changing `locale` to the following gave me a successful build.
```
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

davesrocketshop, 2023-10-03:
I'll look into improving this test

davesrocketshop, 2023-10-06:
I'll have a PR for this later today

davesrocketshop, 2023-10-06:
Well, that was an adventure!

I thought I'd be smart and set the locale in the python so I could test multiple locales. Except the Python objects are created in C++ so that had no effect.

Then I tried changing FreeCAD preferences. Again, no effect. I wasn't going to change my Windows to a locale for a language I don't understand, so over to Linux.

Set my language to de_DE.UTF8 no problem. Fixed the tests. Test with pt_BR.UTF8... problem. For whatever reason, this languages default encoding is not UTF8. The ThermalExpansionCoefficient returns a UTF character (greek letter mu) with its units, so I had to comment out those tests to get it to pass. I'll revisit that later. Tested with pl_PL.UTF8 and it passed fine.

As soon as it passes its build tests I'll create the PR

closed 2023-10-07 by wwmayer