FreeCAD/FreeCAD #10926: [Problem] macOS builds don't include 3Dconnexion support

BenBergman, 2023-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

3Dconnexion support is available for FreeCAD, but on macOS it needs to be available when compiled (https://wiki.freecad.org/3Dconnexion_input_devices#macOS). The official releases don't seem to include this and building from source is not a simple process (I've been attempting it all day without success, and today isn't my first attempt).

It would be great if the official releases had this support enabled.

### Full version info

```shell
[code]
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-10-05:
the official build should include support, what processor are you running and what driver?

you can see here we compile with driver version 10.6.6 for intel and 10.7.0 for arm
https://github.com/FreeCAD/freecad-feedstock/blob/c25b2cc9f13784c94f9d95ae0fe73c11e913f75e/recipe/build.sh#L30-L34

BenBergman, 2023-10-05:
Huh, strange. I've tried a few versions over the past few months and none of them have worked with my SpaceMouse Enterprise.

Processor: 2.4 GHz 8-Core Intel Core i9
OS: macOS 13.5.2 (22G91) (aka Ventura)
[image]

It looks like I'm on 3DxWare 10.7.4. Clicking "Check Update" says I'm up to date, but looking at the website there is a 10.8.1. Either way, it sounds like I'm running a newer version that what is being built into the releases.

adrianinsaval, 2023-10-06:
several people reported that versions beyond 10.7.0 where not working on their macs, so I kept it on that version. I don't own a mac or a spacemouse so I can test myself if newer releases are working, I have to rely on reports of those who do.

For intel, I think we are using the older 10.6.6 release for compatibility with older versions of macOS, I could upgrade this, but I don't really know which minimum version of macOS we should target.

BenBergman, 2023-10-07:
Interesting. Well, maybe I'll try downgrading my 3DxWare version to 10.6.6 then next time I'm at that computer with the SpaceMouse and see how that works.

luzpaz, 2023-10-11:
@BenBergman thanks, please report back. That would be very helpful!

BenBergman, 2023-10-12:
I am back at my computer with the SpaceMouse. I have tried installing 3DxWare 10.6.6 but now my SpaceMouse doesn't even work with the demos. Reinstalling 3DxWare 10.7.0 at least gets the demos working, but FreeCAD still says "No Spaceball Present".

I'm running macOS 13. The main driver page shows 3DxWare 10.8.2 for that macOS version. Looking at the driver archive page there are no listings specifically for macOS 13, but 3DxWare 10.7.0 is officially supported for macOS 10.15, 11, and 12, so maybe it is a safer default than using 3DxWare 10.6.6. https://3dconnexion.com/us/archived-drivers/?fwp_drivers_os=macos-12

I'd be happy to test out a new FreeCAD build with 3DxWare 10.7.0 on my machine.

BenBergman, 2023-10-12:
For some context, Apple no longer supports macOS 10.15 or macOS 11. I think Apple should support their OSes longer, but I suspect most users won't be running macOS 10.14 or older.

adrianinsaval, 2023-10-12:
would 10.7.0 work for you? I can try moving to that driver then if nobody complains we keep it at that. 10.8.2 might not be a good idea if it's dropping support for versions that are still in use (reasonably). I don't want to help apple with planned obsolescence.

adrianinsaval, 2023-10-13:
https://github.com/FreeCAD/freecad-feedstock/commit/f8755d030b75aa704d39cefb6f65baf0afdb9852

latest weekly bundle should be using 10.7.0

BenBergman, 2023-10-13:
Success! So far this is working well for me. Thanks for your help with this, and fingers crossed it doesn't break compatibility for anyone else. 🤞

adrianinsaval, 2023-10-13:
thank you for testing, I'll use this version for the next stable release too.

closed 2023-10-13 by adrianinsaval