FreeCAD/FreeCAD #10927: [Problems] Material system isn't accessible uniformly acrosse the application

sliptonic, 2023-10-03:


luzpaz, 2023-10-03:
Typo in ticket title

yorikvanhavre, 2023-10-04:
I see Material as a module, not a workbench. Like Tux, or MeshPart. It should basically furnish a series of FreeCAD commands, that are to be used in other workbenches, and maybe some sort of direct python functionality, if there is a need for it...

But while it's being built and tested, it might be convenient to have a workbench set up for quickly accessing/testing the different commands...

sliptonic, 2023-10-05:
I still don't know why a workbench would be better at this stage than a module.  A module could still add a toolbar or panel.

yorikvanhavre, 2023-10-09:
Yes, but so there is one centralized place to go look for these commands. But indeed it's a detail.

luzpaz, 2023-10-13:
@sliptonic maybe we should close this in favor of #10174 and move the conversation over there ?

closed 2025-02-17 by github-actions[bot]