FreeCAD/FreeCAD #10928: [Problem] Feed rate missing when using "Z correction" on "Engrave"

fwillo, 2023-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I try to use "Z Correction" on an "Engrave" operation in the Path workbench. Post-processing my operations I get the corrected z-heights. However, on every G1 operation the feed rates are missing. When post-processing the same "Engrave" operation without "Z Correction", I get the output with feed rates.

Please see a working minimal example attached to the issue.

Semi-OT: I usually had working surfaces on non-zero z-heights (e.g. z=1.5 mm in case of the attatched minimal example). Therefore, my probing data was also working on that height (add 1.5mm to the third column of porbeoutput-corr.txt). When applying "Z Correction", the whole output was shifted up to z ≈ 3 mm. So it seems that it was increased by double. Is this intended or maybe not respecting the offset? Also: Is this maybe not the usual approach of creating G-codes in the first place?

Looking forward to your answers.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * freecad.gears 1.0.0
  * Manipulator 1.4.9
[/code]
```

### Subproject(s) affected?

Path

### Anything else?

[working-minimal-example.zip](https://github.com/FreeCAD/FreeCAD/files/12797359/working-minimal-example.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-15:
@fwillo can you test to see that this is reproducible on dev version ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

fwillo, 2023-10-15:
With the weekly-build 
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git)
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * freecad.gears 1.0.0
  * Manipulator 1.4.9
```
from your link I generate an output for which the feed rate is still missing.

PR #11061 (2023-10-16): Add F-word to output on zcorrect dressup
Fixes #10928
commit 7fde4941 (2023-10-16): Fixes #10928
Add F-word to output on zcorrect dressup
closed 2023-10-16 by sliptonic