FreeCAD/FreeCAD #10937: [Problem] Current source doesn't add thumbnail to file with preferences set to save.

macdroid53, 2023-10-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With Save thumbnails set in Preferences, no thumbnail is saved.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34534 (Git)
Build type: Release
Branch: master
Hash: 919f6e78e0222a96a8080298e13a0fe06b59bb08
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.2
  * ZolkoBase 0.0.1
  * lattice2 1.0.0
  * Curves 0.6.13
  * Nodes 0.1.36
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.0
  * Assembly
  * Assembly4 0.50.4
  * freecad.gears 1.0.0
  * sheetmetal 0.3.0
  * Silk 0.1.5
  * kicadStepUpMod 10.19.5
  * A2plus 0.4.60n
  * DynamicData 2.53.0
  * workfeature-macro
  * fasteners 0.4.66
  * ExtremeProDark 2.9.0
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-10-06:
I don't know if it's related but when saving a document the first time then the file dialog pops up. After confirming with Save the application has no active window any more and thus this check fails: https://github.com/FreeCAD/FreeCAD/blob/5a153e50ffb6046b6cc4dba498344032f6edb952/src/Gui/Thumbnail.cpp#L89
What helps is to save the document twice because then the file dialog doesn't pop-up any more and the 3D view remains the active window.

But if the creation of a snapshot fails then as fallback solution it uses the program logo. So, it actually shouldn't happen that the thumbnail is completely empty.

luzpaz, 2023-10-20:
@macdroid53 can you retest and report back ?

luzpaz, 2023-10-23:
@macdroid53 ping

macdroid53, 2023-10-24:
Seems to be ok in: 

[code]
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34771 (Git)
Build type: Release
Branch: main
Hash: d19a16ddd541f8c9cebc12c6d48e8316c150d3a7
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.2
  * ZolkoBase 0.0.1
  * Pyramids-and-Polyhedrons
  * lattice2 1.0.0
  * Curves 0.6.13
  * Nodes 0.1.36
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.0
  * Assembly
  * Assembly4 0.50.4
  * freecad.gears 1.0.0
  * sheetmetal 0.3.1
  * Silk 0.1.5
  * kicadStepUpMod 10.20.7
  * A2plus 0.4.60n
  * DynamicData 2.53.0
  * workfeature-macro
  * fasteners 0.4.67
  * ExtremeProDark 2.9.0
  * Defeaturing 1.2.1
  * CurvedShapes 1.0.4
[/code]

closed 2023-10-24 by luzpaz