FreeCAD/FreeCAD #10939: [Problem]  Part Design groove with Datum lineAxis creates false  error message

edwilliams16, 2023-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On creating the groove the following error message is generated:
`15:06:12  Groove: Revolve axis intersects the sketch
`
which is not the case. However, the groove is correctly executed.
See [https://forum.freecad.org/viewtopic.php?p=711306#p711306](url)

for a sample file and discussion

### Full version info

```shell
[code]
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34478 (Git)
Build type: Release
Branch: master
Hash: 70ca90f46f30c84f9f63d78be6cd14f5a0b93aea
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * DynamicData 2.52.0
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * GDML 1.8.0
  * offline-documentation
  * Manipulator 1.5.0
  * fasteners 0.4.63
  * lattice2 1.0.0
  * CurvedShapes 1.0.4
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.13
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

RohanMishra315, 2023-12-15:
[image]
Check this code , I think the problem arises due to this.
@luzpaz @Roy-043

Roy-043, 2023-12-15:
Thanks. This definitely looks like the right section of the code, but sadly I only know Python.

RohanMishra315, 2023-12-15:
OK ,I'll look into it.

luzpaz, 2025-01-28:
@RohanMishra315 ping