FreeCAD/FreeCAD #10952: [Problem] FreeCAD does not compile (invalid template arguments on constructor declaration)

mwoehlke, 2023-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD does not compile:
```
In file included from freecad/src/CXX/Objects.hxx:42,
                 from freecad/src/Base/PyObjectBase.h:56,
                 from freecad/src/Base/BaseClass.cpp:31:
freecad/src/CXX/Python3/Objects.hxx:1336:21: error: expected unqualified-id before ‘const’
 1336 |         SeqBase<T>( const Object &ob )
      |                     ^~~~~
```

Although the constructor declarator `class_name<T>(...)` is accepted by many compilers, **it is not valid C++**, and it's unclear that it was *ever* valid C++. GCC ≥ 11 in `-std=c++20` mode or later rejects it (albeit with a rather confusing error message, as above).

See also [CWG Issue 2237](https://cplusplus.github.io/CWG/issues/2237.html).

### Full version info

```shell
master circa 2023-10-04
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwoehlke, 2023-10-06:
I'll try to provide a PR for this in the next few days (it's a relatively easy fix, although a number of places in that file are affected), however, it's unclear if this file is hand-written or generated. If it's generated, I don't know where the code generation happens. (I'll also try to remember to update the issue description with the affected SHAs; my checkout is on a different computer...)

wwmayer, 2023-10-06:
> it's unclear if this file is hand-written or generated

It's hand-written. It's part of the PyCXX project which is hosted on Sourceforge: sourceforge.net/projects/CXX

wwmayer, 2023-10-06:
In upstream it's already fixed: https://sourceforge.net/p/cxx/code/449/tree//trunk/CXX/CXX/Python3/Objects.hxx?diff=517694e95fcbc9797b11d5c7:448

mwoehlke, 2023-10-07:
Slightly better link: https://sourceforge.net/p/cxx/code/449/

...and, yup, that looks like exactly the changes I made. Is it useful if I submit that as an independent PR, or do you have a better handle than I do how to go about updating that particular code?

wwmayer, 2023-10-07:
If you have it ready you can make a PR. Otherwise I will do it in the next couple of days.

PR #10970 (2023-10-09): Fix build for C++20
Remove template argument list from constructor names. It has never been necessary, and is forbidden in C++20 (n.b. [CWG 2237](https://cplusplus.github.io/CWG/issues/2237.html)). This has also been independently fixed upstream in [r449](https://sourceforge.net/p/cxx/code/449/).

Fixes #10952.
commit a7e4804e (2023-10-09): Fix build for C++20

Remove template argument list from constructor names. It has never been
necessary, and is forbidden in C++20 (n.b. CWG 2237¹). This has also
been independently fixed upstream in r449².

Fixes #10952.

¹ https://cplusplus.github.io/CWG/issues/2237.html
² https://sourceforge.net/p/cxx/code/449/
closed 2023-10-09 by wwmayer