FreeCAD/FreeCAD #10965: Part Design Workbench: ShapeBinder UI weirdly bad

your-friend-alice, 2023-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the UI for adding geometry to a ShapeBinder, you have to activate the "Add Geometry" button, and then you can click on a single line, and then you have to go back and re-click "Add Geometry" for every single line segment. This gets old fast. Also, if you highlight some geometry and then click "Add Geometry", your selection is ignored and cleared. I noticed the Chamfer UI has very similar functionality (adding geometry to a list) but works much better with its "Select" mode.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (bspwm)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

your-friend-alice, 2023-10-31:
oh, good call on the edit, just a note on the old title's intent, the point of "weirdly bad" was to emphasize that its badness was exceptional, since freecad seems to represent this type of operation quite a bit better in other contexts. new title is less potentially rude and more descriptive so no objections here!

luzpaz, 2023-10-31:
It's better to be as neutral as possible when it comes to working with volunteers and in manners of taste and perspective.

closed 2025-02-17 by github-actions[bot]