FreeCAD/FreeCAD #10967: Sketcher: add customizable tags to lines

your-friend-alice, 2023-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A Part Design workflow I really like is to make a single "main" sketch for each plane, and then use ShapeBinders to pull out different subsets of the geometry from the sketch to use for different features. However, for complex sketches it's a pain to manually add all the geometry, and the work is multiplied by the number of different ShapeBinders and the number of times you go back and make any sort of non-trivial edit to the sketch.

What would be nice is if I could explicitly assign "tags" to geometry in a sketch. Then I could assign the lines making up the outline the "outline" tag, give some circles a "mounting_holes" tag, and add other tags like "pocket_shallow", "speed_holes", etc to reflect the intended use of the geometry. Then, when creating a ShapeBinder, instead of just manually selecting lines from a sketch, provide an option to supply a boolean tag matching expression. like, a ShapeBinder for the Pad of a part might just have an expression of `outline`, but you could do more complex things like "tapped_hole && !big".

This still leaves the annoying task of assigning tags to each element in a sketch, but:
- this can be helped a lot with a decent UI
- it can easily remain optional by preserving the existing behavior of ShapeBinders and either making the tag matching expression an alternate mode for the ShapeBinder, or creating a new type of object to represent this operation
- It's tedious manually selecting geometry to apply tags to, but that's true with the status quo where you have to manually select geometry for the ShapeBinders. The benefit of tags in the sketcher would be, you change something in one place, appropriately tag any new geometry once, and the change automatically propagates to all the features affected.
  - This also nicely sidesteps a lot of topological naming issues. Often when modifying geometry in the Sketcher, the ShapeBinders' selections become basically randomized, and I end up having to redo ShapeBinders that have nothing to do with the changes I made (since the indices get shuffled around when adding/removing lines). If using a tag matching expression, the ShapeBinder no longer cares about edge indices as a stable reference, since it can just recompute those whenever something changes. Also, the geometry can use the tags as part of their ID, which would help stabilize the identities of edges further down the line as well.

I think generally, allowing the user to associate custom metadata with lines in a Sketch has a ton of potential for other tools to make use of, far beyond what I'm even considering in this request.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (bspwm)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-10-09:
This type of functionality is usually known/implemented as _layers_

adrianinsaval, 2023-10-09:
@abdullahtahiriyo

your-friend-alice, 2023-10-09:
Layers would be helpful, and would benefit my workflow, but it falls way short of what tags can provide. My understanding of what people mean when they say layer is, "a layer is a set of objects, each object can only be a member of exactly 1 layer". Also, it seems that there's typically a single fixed relationship between the layers (they have a "stacking order" which is used in most operations). Meanwhile, tags have far fewer restrictions. Like layers, tags are sets, but objects can be a member of any combination of tags, and the concept of tags doesn't have any implicit ordering or relations between the tags, it's entirely up to the user/application to define the behavior of the tags using boolean logic.

Here's a contrived example: with layers, "select all objects on the 'Screw' layer, that aren't on the 'Countersunk' layer" is a nonsensical query because by definition all objects on the "Screw" layer are not in the "Countersunk" layer. Instead, you need to create "Countersunk Screw" and "Flathead Screw" layers, and hope that the implementation lets you do arbitrary unions of layers if you need to perform a common operation on all screws.

And even if it does, that's not a perfect solution, because tags deal with cardinality much better than layers. Imagine you have multiple variations of a screw hole you want to be able to select on, you have "countersunk vs flat head" screws, you have "threaded vs clearance" holes for the screws, you have "precise vs loose", and you have "M3, M5, M8" sizes. To represent that space, you're going to need to create 24 layers!

Even with something much smaller like, "chamfered vs non-chamfered, outline vs pocket edges", the cardinality is manageable, but the structure of an explicit layer for each combination is just awkward to work with.

Anyways. Layers would be cool too, it would certainly be better than the current state, but this request is for something distinct from layers.

adrianinsaval, 2023-10-09:
layer groups could handle some of what you're saying. Beyond that, I have doubts that it's really worth the effort to implement tags, the usecase seems niche.

keep in mind that making very complex single sketches and deriving multiple features from it is discouraged since having too many elements on a sketch slows down the solver. Instead I would encourage you to create multiple more simplistic sketches.

abdullahtahiriyo, 2023-10-13:
I have to look into how Shapebinders actually work. I am unsure if they work based on sketch geometry or sketch shape.

The initial idea of Sketch Layers was that each geometry can be in just one Layer. However, it is perhaps possible to allow a mode were a geometry can be in more than one layer. I have not evaluated all the consequences yet.

I was thinking on using layers to enhance carbon copy. There, one could carbon copy several layers of a sketch (where geometry elements are just in one layer), so there might be a first layer "minimal master sketch" giving all the dimensions on which other layers build, and then different layers to develop parts. Then, one could carbon copy the "minimal master sketch" together with other developments. Then I also thought that this layering could help in the synchronisation when having to alter the master sketch. But, of course, this is still not properly thought through...

I take the input for when I develop this farther...

adrianinsaval, 2023-10-13:
> I have to look into how Shapebinders actually work. I am unsure if they work based on sketch geometry or sketch shape.

they use the shape