FreeCAD/FreeCAD #10971: [Problem] Uninstaller on Windows doesn't work when installed for current user

alexbeer2048, 2023-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On Windows, if you choose the option to install only for current user when installing FreeCAD, the setup finishes normally and the app also works fine. So far so good.

But if you try to uninstall the application in the Apps & Features tab of the Windows settings it does not work as intended. All files, links, file associations, etc. get removed but the application entry will not. It will leave a "broken installation" which can not be removed.

The next attempt to uninstall it will give an error message that the "....../Uninstall-FreeCAD.exe" could not be found.

### Full version info

```shell
FreeCAD 0.21.1 (WIN x64 release versions)
FreeCAD 0.21.0
FreeCAD 0.20.2
I haven't tested older versions than that.

Windows 11 Version 22H2
```

### Subproject(s) affected?

None

### Anything else?

Here are screenshots of broken app entries and the error message I get if I try to (re-)uninstall an already uninstalled version of FreeCAD: (sorry it's in German)
[image]

[image]

I tried to analyze a little bit what could be going on and my findings so far are:

Looking at the Windows registry, before and after installing and uninstalling.

Install **for all users** (option): Creates new entries in 
"HKEY_LOCAL_MACHINE\SOFTWARE\FreeCAD0211" "HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall\FreeCAD0211"
Uninstaller: Removes these two entries

Install **for current user only** (option): Creates new entries in 
"HKEY_CURRENT_USER\Software\FreeCAD0211" "HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Uninstall\FreeCAD0211"
Uninstaller: Leaves these registry keys (and contained values) as is. --> So it leaves the app entry for Windows

So my best guess is that the uninstaller doesn't look for all the right keys to remove if FreeCAD was installed for the current user only.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-08:
We should probably backport a fix for this. @chennes what do you think ?

luzpaz, 2023-10-13:
> Install **for current user only** (option): Creates new entries in "HKEY_CURRENT_USER\Software\FreeCAD0211" "HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Uninstall\FreeCAD0211" [...]

where is the code for this ? (Libpack ?)

chennes, 2023-10-13:
No, the NSIS setup is in the tools directory.

alexbeer2048, 2023-10-14:
After some digging deeper (and brushing up on my NSIS skill) I found something that might be the fix.

Looking at the doc of the MultiUser library: https://nsis.sourceforge.io/Docs/MultiUser/Readme.html

There is a link to an example script at the bottom of the page and there it has a line:

```nsis
Section
  ...
  WriteRegStr ShCtx "${UNINSTKEY}" $MultiUser.InstallMode 1 ; Write MULTIUSER_INSTALLMODE_DEFAULT_REGISTRY_VALUENAME so the correct context can be detected in the uninstaller.
  ...
SectionEnd
```

So basically InstallMode should also be written to the registry within the install section of the script. Not sure why this is hidden in the example, but that's what stood out to me compared to what we have right now.

In "FreeCAD/src/WindowsInstaller/setup/configure.nsh"
I would put this right after [line 53](https://github.com/FreeCAD/FreeCAD/blob/main/src/WindowsInstaller/setup/configure.nsh#L53):

```nsis
  WriteRegDWORD SHCTX ${APP_UNINST_KEY} "NoModify" 0x00000001
  WriteRegDWORD SHCTX ${APP_UNINST_KEY} "NoRepair" 0x00000001
  WriteRegStr SHCTX ${APP_UNINST_KEY} "StartMenu" "$SMPROGRAMS\$StartmenuFolder"
  <---here
```

Not sure if this would do the trick or if something else needs attention.

When I have time I will try to set up the build pipeline so I can build the installer package on my machine and test if it works.

Until then, maybe someone who has more experience with the installer scripts than me can take a look at this and tell me if this makes sense or I'm on the wrong track.

RohanMishra315, 2023-10-17:
hey @alexbeer2048 @luzpaz i think this could help , To ensure that registry keys related to FreeCAD are removed for the current user installation, we can add a conditional check to target the current user's registry entries when uninstalling

Section "un.FreeCAD" un.SecUnProgramFiles
  SectionIn RO
  # ...
  ${if} $MultiUser.InstallMode == "CurrentUser"
    DeleteRegKey HKCU "Software\Classes\${APP_EXT}"
    DeleteRegKey HKCU "Software\Microsoft\Windows\CurrentVersion\Explorer\FileExts\${APP_EXT}"
    # Continue with other registry key removals specific to the current user installation
  ${endif}
  # ...
SectionEnd

The if condition checks whether the installation mode is set to "CurrentUser," and if so, it targets the current user's registry entries for removal. What you think

alexbeer2048, 2023-10-18:
> hey @alexbeer2048 @luzpaz i think this could help , To ensure that registry keys related to FreeCAD are removed for the current user installation, we can add a conditional check to target the [...]

@RohanMishra315 thank you for your input. I'll test if this can also help fix the problem.
I just managed to set everything up in order to create the installer package. (Maybe I could have skipped the compile the app from scratch step and copied the files from an existing installation, but I regret nothing :P )

I'm just at the beginning, but already realized that debugging is not a big feature in the NSIS language.

**Tips on** how to set up some decent **debugging** for NSIS scripts are very welcome!!!

For now, MessageBox is my new friend.

luzpaz, 2023-10-20:
@alexbeer2048 keep us posted :wink:

alexbeer2048, 2023-10-20:
@luzpaz I found the culprit(s) :see_no_evil:

In the `un.onInit` section we are using the `SetShellVarContext` after the `MULTIUSER_UNINIT` macro.

https://github.com/FreeCAD/FreeCAD/blob/ce9c57383a026aad024ca4cf8a489f066496f175/src/WindowsInstaller/include/init.nsh#L178-L184

Here it's hidden inside the macro :smiling_imp:
https://github.com/FreeCAD/FreeCAD/blob/ce9c57383a026aad024ca4cf8a489f066496f175/src/WindowsInstaller/include/init.nsh#L203-L204

This undoes what the `MULTIUSER_UNINIT` macro is supposed to do, which is to determine and set the correct shell context according to the option the user selected during installation.

So the SHCTX root key variable we're using to write and delete reg keys is not necessarily the same anymore for the uninstaller compared to the installer.
This means for example we would write to _HKEY_CURRENT_USER_ during installation and then try to remove these keys from _HKEY_LOCAL_MACHINE_ during uninstall.

Working on the fix right now!

The compile time for the installer on my old laptop is slowing me down quite a bit, so bear with me.

Testing will also take some time, to make sure everything works in the different possible scenarios (all user/current user option, as admin/standard user) and also that I don't break things.

luzpaz, 2023-10-23:
thanks for all your work on this @alexbeer2048

RohanMishra315, 2023-10-24:
Yes that's great work @alexbeer2048 . Also @luzpaz @alexbeer2048 could you tell where I should engage with you all (community server) . I have lot's of questions to ask , tell with where we can discuss . Mind me , for asking all these here.

alexbeer2048, 2023-10-27:
Update: almost done, but still working on some minor issues I want to improve before I release this into the wild.

I'm planning on wrapping up and making the PR over the next week. Time to start reading the contributing guidelines:smile:

PR #11268 (2023-11-14): Fix windows uninstaller issue #10971 and #11232
First PR, so happy for any advice on how to do this properly.

Fixed #10971 

Also changed FileDescription for Windows installer/uninstaller for #11232
from "FreeCAD - Your Own 3D Parametric Modeler" to "FreeCAD Installer/Uninstaller"
commit 63737608 (2023-11-14): Fix windows uninstaller issue #10971 and #11232 (#11268)

* Move MultiUser install mode default registry key to the uninstaller registry entry

Lets the uninstaller detect the correct context for each installed version. related to #10971

* Fix uninstaller doesn't use correct shell context when installed for current user

related to #10971

* Update installer init checks for previously installed versions

Use shell context determined by the MultiUser plugin. related to #10971

* fix: Perform installer version checks after install mode selection

Perform installer version checks after install m [... cut]
closed 2023-11-14 by adrianinsaval