FreeCAD/FreeCAD #10974: [Problem] Errors on access of QtGui in python console

alexneufeld, 2023-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Paste the following into the python console:

``` python
from PySide import QtGui

```
Now *type* the following in, *one chacter at a time*:
``` python
QtGui.Q
```

This causes the following error:

``` bash
/tmp/.mount_FreeCAgAysuy/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/signature/parser.py:200: RuntimeWarning: pyside_type_init:_resolve_value

        UNRECOGNIZED:   'int64_t*'
        OFFENDING LINE: 'PySide2.QtGui.QOpenGLExtraFunctions.glGetBufferParameteri64v(self,target:unsigned int,pname:unsigned int,params:int64_t*)'
        
  warnings.warn("""pyside_type_init:_resolve_value
/tmp/.mount_FreeCAgAysuy/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/signature/parser.py:200: RuntimeWarning: pyside_type_init:_resolve_value

        UNRECOGNIZED:   'int64_t*'
        OFFENDING LINE: 'PySide2.QtGui.QOpenGLExtraFunctions.glGetInteger64i_v(self,target:unsigned int,index:unsigned int,data:int64_t*)'
        
  warnings.warn("""pyside_type_init:_resolve_value
/tmp/.mount_FreeCAgAysuy/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/signature/parser.py:200: RuntimeWarning: pyside_type_init:_resolve_value

        UNRECOGNIZED:   'int64_t*'
        OFFENDING LINE: 'PySide2.QtGui.QOpenGLExtraFunctions.glGetInteger64v(self,pname:unsigned int,data:int64_t*)'
        
  warnings.warn("""pyside_type_init:_resolve_value

```

Note that this error only occurs once per session, `QtGui` still seems to work normally.

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34542 (Git) AppImage
Build type: Release
Branch: master
Hash: d72478ade26e163d1429a8c152db45df5e96106c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * Curves
  * python_site_packages
  * Pyramids-and-Polyhedrons
  * FreeCAD-Flat-Icons
  * Assembly3 0.12.0
  * OpenDark 2023.10.2
  * robotpy
  * brlcad_wb
  * fasteners 0.4.62
  * FreeCAD-PreferencePacks.backup1669700355.378573
  * Behave-Dark-Colors 0.1.1
  * freecad.gears 1.0.0
  * Assembly4
  * FreeCAD_SheetMetal 0.2.56
  * Silk
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-10-09:
I cannot reproduce it with:

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34553 (Git)
Build type: Debug
Branch: master
Hash: 5f14ef76566be011b3b7b2d1a18ec11c40763585
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34594 (Git)
Build type: Release
Branch: master
Hash: e9187412d9b3b372ee4c1d7e2fc7a0bf08a2e647
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

----

I can reproduce it with:

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34310 (Git) AppImage
Build type: Release
Branch: master
Hash: 3e789dfc85c84e5e6295df908795db5830d28d43
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

The problem seems to come from _AppImage_.

alexneufeld, 2023-10-10:
i can also produce the issue with the following:

``` ruby
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.5, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/Canada (en_CA)
```

bdieterm, 2023-10-13:
That seems to be a problem inside the package for PySide2 and Shiboken2.

I can reproduce the problem using an extracted AppImage by executing the line
`from PySide2 import QtGui; s = QtGui.QOpenGLExtraFunctions.__doc__`
even when doing this in `./usr/bin/python` from the AppImage.

As a workaround, execute the following lines in FreeCAD's Python interpreter:
```
import shiboken2
from shibokensupport.signature.mapping import type_map
type_map['int64_t*'] = type_map['long*']
```
and then there are no warnings anymore.

luzpaz, 2023-10-15:
@hfiguiere would you mind trying to reproduce this on FreeCAD flatpak ?

luzpaz, 2023-10-31:
@adrianinsaval can you reproduce on flatpak ?

hfiguiere, 2023-11-01:
```python
from PySide2 import QtGui; s = QtGui.QOpenGLExtraFunctions.__doc__
```

This doesn't produce any error.

bdieterm, 2023-11-01:
As I wrote 3 weeks ago, the problem seems to be inside the package for PySide2 and Shiboken2.

The current FreeCAD weekly build AppImage gives the following version information:
```
>>> import shiboken2
>>> shiboken2.__version__
'5.15.8'
>>> import PySide2
>>> PySide2.__version__
'5.15.8'
```

There is no PySide2 version 5.15.8 on pypi.org so I will use a fresh miniconda installation to reproduce the problem:
* download miniconda from https://repo.anaconda.com/miniconda/Miniconda3-py310_23.9.0-0-Linux-x86_64.sh
* install it via `bash Miniconda3-py310_23.9.0-0-Linux-x86_64.sh`
* install PySide2 via `conda install -c conda-forge pyside2=5.15.8`
* run `python3.10` from the new miniconda installation
* execute python code `from PySide2 import QtGui; s = QtGui.QOpenGLExtraFunctions.__doc__`
* --> same error as initially reported (but no FreeCAD involved)

So, I would say, this is not an AppImage problem but rather a problem of the [PySide2/Shiboken2 package that is installed via conda](https://anaconda.org/conda-forge/pyside2).

closed 2025-01-27 by alexneufeld