FreeCAD/FreeCAD #10975: Constraints removal in selection box for parabole and hyperbole branch sometimes crashes FC.

Erik84750, 2023-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Please refer to FreeCAD forum thread https://forum.freecad.org/viewtopic.php?t=81580 

The issue concerns the inability to remove more then one selected constraint in the selection box, and for hyperole and parabole branch this results in -sometimes- FC crash.
Abdullah is in the process of investigation.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Belgium (nl_BE)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-12-02:
Dear @abdullahtahiriyo, [this commit](https://github.com/FreeCAD/FreeCAD/compare/main...xtemp09:FreeCAD:iss_10975) seems to fix the issue. Could you take a look at it?

I made the commit without going deep into it, but the crash occurs because the notification area functions try to retrieve the name of the deleted object, causing segfault.

abdullahtahiriyo, 2023-12-06:
Please, PR that.

xtemp09, 2023-12-09:
This issue should be closed. 👍

closed 2023-12-09 by abdullahtahiriyo