FreeCAD/FreeCAD #10976: Hidden lines in exported PDF are not visible, because the scale is to small

bewegende-Architektur, 2023-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hey there,

it looks like there is an issue with the scale of lines in exported PDF.
https://forum.freecad.org/viewtopic.php?p=711669#p711669

There has not been an issue here on github, so hopefully it is ok that i have created it.

I am facing the same issue on multiple machines running manjaro.
App, snap and community versions have been tested.
Is there any workaround like changing qt settings / config for the line spacing?

Thanks in advance.

Greetings,
Chris

FreeCAD 0.21.1, Libs: 0.21.1R33694 (Git)

6.1.55-1-MANJARO #1 SMP PREEMPT_DYNAMIC
Sat Sep 23 12:13:56 UTC 2023 x86_64 GNU/Linux

Qt 5.15.10 (x86_64-little_endian-lp64 shared (dynamic)
release build; by GCC 13.2.1 20230801) on "xcb"

### Full version info

```shell
[code]
OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.5, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.4.60
  * Glass (Disabled)
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-10-09:
How are you making the PDF?  There is the "export pdf" method in FreeCAD and also the "print to file" in the OS.    The two methods use different code (AFAIK).

bewegende-Architektur, 2023-10-10:
> How are you making the PDF? There is the "export pdf" method in FreeCAD and also the "print to file" in the OS. The two methods use different code (AFAIK).

Thanks for your response. I have tried both and printing on paper too. I am using Freecad mostly for modelling. Therefore i have not recognized this issue earlier, but i can remember that i have exported succesfully some time ago. I have tried earlier version to but with the same problem. So i guess it is more a problem in my current setup of QT or something else similar.

WandererFan, 2023-10-11:
Can you see the hidden lines in this pdf?  

[HiddenLines.pdf](https://github.com/FreeCAD/FreeCAD/files/12869822/HiddenLines.pdf)

luzpaz, 2023-10-11:
@WandererFan here's what i see when I open HiddenLines.pdf

#### Firefox 117.0.1 (Manjaro Linux)

[image]

#### Okular Version 23.08.1 (Manjaro Linux)

[image]

bewegende-Architektur, 2023-10-12:
[image]

Thank you very much for your time.

Yes, your PDF looks similar to my Setup in Manjaro.
The hidden lines are drawn, but way to small.

bewegende-Architektur, 2023-10-12:
Yes, the thickness is influencing the export. Here are some examples:
[hidden_0-13.PDF](https://github.com/FreeCAD/FreeCAD/files/12883863/hidden_0-13.PDF)
[hidden_0-50.PDF](https://github.com/FreeCAD/FreeCAD/files/12883865/hidden_0-50.PDF)
[hidden_0-80.PDF](https://github.com/FreeCAD/FreeCAD/files/12883866/hidden_0-80.PDF)
[hidden-line_test.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/12883940/hidden-line_test.FCStd.zip)

Thanks for pointing this out. It looks like the dash-size is connected to the thickness.
Is it possible to adjust thickness and dash-size separately?
Would it be possible to change the dash-pattern to make dash and space the same size?

WandererFan, 2023-10-12:
We are currently using the dash patterns supplied by the Qt framework.  We don't have control over those patterns.

We are working on #9230 which will use the dash patterns specified in ISO128-20.   Note that in ISO128-20 the lengths of the marks and spaces are functions of the line width.

bewegende-Architektur, 2023-10-12:
[image]

Ok, i see. Thank you for pointing this out. If the lines in exported PDF are written by QT, how are the lines in viewport drawn? They are scaled correctly from my point of view.

bewegende-Architektur, 2023-10-12:
Oh wait, maybe they are the same, when looking closely.

Thanks for your time :)

WandererFan, 2023-10-12:
The screen display and the printed page are both rendered by Qt, but the resolution of the display devices will be different.  Maybe that has something to do with it.

WandererFan, 2024-02-14:
This is fixed in v0.22 with the new standards compliant line processing.

closed 2024-02-14 by WandererFan