FreeCAD/FreeCAD #10982: [Problem] Cannot set any of the constraints in Sketcher on macOS

tamasgal, 2023-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I just updated from 0.19 to 0.21.1 since the older version crashed very often.

0.21.1 is totally unusable since e.g. in sketch mode, none of the constraints can be displayed or edited. Also all the symbols are super tiny (working on a 4K monitor).

I already deleted all the preferences and cache from the previous installations, so basically started with a fresh and clean install but I cannot use Sketcher at all.

[image]

[image]

### Full version info

```shell
[code]
OS: macOS 13.3.1
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tamasgal, 2023-10-09:
OK thanks, that helped!

cbenhagen, 2024-06-04:
@adrianinsaval can we re-open this issue? The problem still persists. A new user shouldn't be expected to go into settings and change the font size. A sane default should be used. Maybe this in fact not a settings issue but a HiDPI issue?

tamasgal, 2024-06-04:
I agree that it's not the best UX. Another issue which I still have is that I switch between high and low resolutions, so I need to toggle this each time.

cbenhagen, 2024-06-04:
@tamasgal can you reopen the issue?

tamasgal, 2024-06-07:
I just realised that the TechDraw is also completely messed up. I don't even know which setting to adjust to make the connection points smaller:

[image]

xtemp09, 2025-01-12:
_Qt_ programs can be scaled. @tamasgal, could you launch _FreeCAD_ with an environment variable set? I've just [written a commentary](https://github.com/FreeCAD/FreeCAD/issues/16652#issuecomment-2585711944), perhaps, it is inappropriate to double-post.