FreeCAD/FreeCAD #10985: importDXF does not import DXF properly. ARC and CIRCLE are placed at the wrong location]

ndl303, 2023-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Thanks for the great product.

I have a relatively simple (R2000) DXF file that has a single 3D component, (the part was created by python package ezdxf.py which read the part from an old  Cadkey97 export file). The part consists of 164 LINES, 62 ARCS and 27 CIRCLES.  There are no LAYERS, BLOCKS etc.  I have tried both the default import DXF and the legacy import DXF and I have played with a few of the import options. But I have had no success. All the LINE entities are imported correctly but the CIRCLE and ARC are incorrectly drawn in the workspace, see  figure below,  

[image]

It looks like the circles and arcs have been reflected/projected/directed to the 4 corners of the image rather than being placed close to the central square object

I have imported the same DXF file into Blender and it has  no issue rendering the object correctly; all the circles and arcs are in the correct place, see below
[image]

and I have also imported the same file into E-drawings and it has no issue rendering the object correctly, again all the circles and arcs are properly located, see below.
[image]

I am new to FreeCad and I am still very green with the whole environment. However, I would guess that you have improperly implemented the Autocad OCS (Object Coordinate System) used by CIRCLE, ARC and other planar objects. I had a quick look inside `importDXF.py `and I can see that the code is aware of the OCS and it definitely looks like you are doing the right sort of thing. But I am not sufficiently knowledgeable on Freecad (or DXF) to verify that it is the correct transformation. Given that it seems to be the "odd-one-out" I suspect  there is a problem lurking somewhere inside the importDXF transformations.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Draft

### Anything else?

I have attached the DXF file. I have tried to keep it very simple.
[test.zip](https://github.com/FreeCAD/FreeCAD/files/12851847/test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-10-10:
In librecad it also does not open correctly:
[image]
There is probably something wrong in the file, but I'll have a better look

ndl303, 2023-10-10:
Thanks for the quick response. Interesting! I will also take a closer look and perhaps look a test case in some detail. Perhaps the file I provide has an issue that blender and E-drawing “fix” but other DXF implementations don’t. 

 

 <https://user-images.githubusercontent.com/1136856/273889659-15aad154-3f90-4c6c-90e5-254cbf199360.png>

Roy-043, 2023-10-10:
The DXF is fine. LibreCAD is not 3D capable.

FreeCAD has 2 DXF importers: the 'new' C++ importer and the legacy importer. If you select the legacy importer and specify that Draft objects should be created, the file imports correctly (except that the last object gets a different color for some reason). See https://wiki.freecad.org/Import_Export_Preferences#DXF.

The C++ importer completely ignores the DXF OCS AFAICT.

The legacy importer ignores the DXF OCS if Part shapes are created. I'll do a PR to fix that.

ndl303, 2023-10-11:
Thanks Roy-043,

 

That almost worked the first time! The decoder was dropping arcs whose end angle was greater than 360 degrees.

 

I subtracted 360 degrees from the start and end angle of these problem arcs, regenerated the DXF and now it works as expected.

 

 

Many thanks for the assistance.

ndl303, 2023-10-11:
Whoops, I missed the images.

This is the first image with missing arcs
[image]

and after subtracting 360 from the start and end angle of the missing arcs I get

[image]

closed 2023-10-11 by ndl303

ivor108, 2024-02-07:
Hello, I have a similar problem. But when I use the legacy importer I got this error:
File "<string>", line 1, in <module>
  File "/app/freecad/Mod/Draft/importDXF.py", line 2777, in insert
    processdxf(doc, filename)
  File "/app/freecad/Mod/Draft/importDXF.py", line 2390, in processdxf
    shape = drawSpline(spline)
            ^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/Draft/importDXF.py", line 1556, in drawSpline
    return Draft.make_bezcurve(points, Degree=degree)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'TypeError'>: make_bezcurve() got an unexpected keyword argument 'Degree'