FreeCAD/FreeCAD #11003: [Problem] PartDesign boolean with SubShapeBinder reports "The graph must be a DAG"

Anthony-Gaudino, 2023-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After doing a PartDesign boolean union with two bodies, where one body contains a SubShapeBinder of the other body, FreeCAD reports the `The graph must be a DAG` error.

How to reproduce:
1. Create **partA** with a solid cube.
2. Create **partB** with a solid and a SubShapeBinder referencing an edge from **partA**
3. Select **partA** and boolean union with **partB**
4. `The graph must be a DAG` error appears on report view

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.8.18
  * Silk 0.1.5
  * Defeaturing 1.2.1
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.66
  * lattice2 1.0.0
  * sheetmetal 0.3.1
  * toSketch 1.0.1
  * Curves 0.6.13
  * Assembly4 0.50.4
  * Assembly3 0.12.0
```

### Subproject(s) affected?

PartDesign

### Anything else?

It seems that the error doesn't happen with ShapeBinder.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-10-11:
Confirmed. The SubShapeBinder references "Body [Box.Edge6]", the ShapeBinder references "Box [Edge6]". I am not sure why that is.

To avoid the problem you can edit the Support property of the SubShapeBinder:

1. Right-click in the Tree view and select "Show all".
2. Right-click in the "Support" field and uncheck "ReadOnly".
3. Click in the "Support" field.
4. Click the "..." button.
5. Make a note of the edge number.
6. Click the "Clear" button.
7. Select the unnested Box from the list.
8. Double click in the cell to the right of it.
9. Enter the edge number for example: Edge6.
10. Reselect the unnested Box.
11. Click "OK".

adrianinsaval, 2023-10-16:
@realthunder could you shed some light here?

luzpaz, 2025-09-19:
@Anthony-Gaudino is this still the case for the dev build ?

FEA-eng, 2025-09-20:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-10-08:
Still relevant (note that partA and partB are just Bodies):

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261006 (Git shallow)
Build date: 2026/10/06 21:00:40
Build type: Release
Branch: (HEAD detached at 3424c4106)
Hash: 3424c410627263394f701aeec5b61d7bab7bb3a1
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```