FreeCAD/FreeCAD #11004: [Problem] Huge size of the preferences dialog

wwmayer, 2023-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the merge of the PR https://github.com/FreeCAD/FreeCAD/pull/7888 the preferences dialog now occupies almost the full height of the screen. This is caused by the DlgSettingsUI because it contains more than 30 controls in a vertical layout.

This huge height affects all other preference pages and for most of them there is now a blank area and at the bottom are standard buttons. This leads to a pure usability because one has to move the mouse like crazy in order to accept the changes.

Either the UI tab should be split into two tabs or the whole page should be embedded into a scroll area.

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34633 +4 (Git)
Build type: debug
Branch: overlay_fixes
Hash: 31e8c2c80b01b17f9a55a06dd20b9c5d0e14d7ad
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-10-11:
I think a re-organisation of the page (using 2 column for a start) will help.
Besides many settings seems very advanced. I think some could probably be removed from the preference page. 
Or perhaps at least a 'Show advanced / Hide advanced' button that shows/hides most of these.

PaddleStroke, 2023-10-12:
Hmm so several things : 
- 'Tree view' could be organized like this to save space: 
[image]

- 'Overlay' has many preferences that should not be exposed to user at all. It feels like a theme developper dashboard. I think most of them would be fine to be accessible only through parameters and not in preferences.
[image]

- The UI preference page is not written in a .ui file, but in a python file. I'm not sure what we should do. Should we write a .ui file so that it is the same as other preference pages rather than using this new python template ? Because this python template seems to generating only single rows.

wwmayer, 2023-10-12:
> Overlay' has many preferences that should not be exposed to user at all... I think most of them would be fine to be accessible only through parameters and not in preferences.

Having it only in the parameter editor might be a bit cumbersome because it will be more complicated to figure out the right configuration. But we can put a button "Advanced..." into the preference page and move all these settings into a new dialog.

PaddleStroke, 2023-10-12:
Note: For the advanced button I was thinking about just showing / hidding a groupbox rather than an independant dialog.

@chennes that was my feeling. Now that they here its like, why not leaving them in the advanced section. But if we do that everywhere in the application we're going to have thousands of settings.

wwmayer, 2023-10-12:
> Note: For the advanced button I was thinking about just showing / hidding a groupbox rather than an independant dialog.

But then we have the same problem again that the preferences dialog becomes huge -- at least tmp. As Chris said these are rather esoteric settings and it isn't a big issue if they are not directly accessible.

luzpaz, 2023-10-12:
Related: #10886

maxwxyz, 2023-11-24:
The preferences dialog is too large on my screen, so it is not possible to click "OK" or "CANCEL" nor reach the "Reset..." group.
The dialog cannot be resized, when I try to scale it vertically, the dialog moves quickly out of the screen (into the bottom) and is unable to reach again. I can only close it with ESC.

Quick Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/0d61ff61-bbc0-4821-b548-e3745183523d

### Full version info

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35176 (Git)
Build type: Release
Branch: main
Hash: dcd533af4bf0cb3dbc50c7d0e2fb579cad168819
Python 3.10.13, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.8
  * CurvedShapes 1.0.5
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.70
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.12
  * woodworking 0.21.30486
```

PaddleStroke, 2023-11-24:
This is a double effect of both #7888  and #11018
7888 added a super huge pref page that needs to be trimmed
11018 apparantly removed the possibility to resize the pref window

wwmayer, 2023-11-25:
> The preferences dialog is too large on my screen, so it is not possible to click "OK" or "CANCEL" nor reach the "Reset..." group.

This is a regression of #11018. All pages used to be embedded into a scroll area so that it was guaranteed that the dialog showed the standard buttons. This scroll area has been removed for no obvious reason.

kadet1090, 2023-11-26:
>This scroll area has been removed for no obvious reason.

Scroll area was removed as it was implemented in a way that made it a lot harder to reason about the UI state. On my side I did not notice any regression as my window manager for some reason allows resizing windows that should not be resizable. But yeah, I wasn't my intent to remove it but to simply replace it with static widget. 

It is revived in the #11540 which also changes default height of the preferences dialog to 900px so huge pages should no longer impact the modal window size and should be simply scrollable by default.

closed 2023-11-27 by chennes

PaddleStroke, 2023-11-29:
As a note, I'm offering some changes that would rework this huge UI page by simply splitting it in TreeView and Overlay : 
https://github.com/FreeCAD/FreeCAD/pull/11581

PR #11540 (2023-11-27): Gui: Add ScrollArea to Preferences and auto expand groups
This adds QScrollArea widget to DlgPreferences which in turn allows preference pages to be scrollable. Stacked widget sizes are forced to selected widget so scroll area should appear when it is needed.

Second commit ensures that after selecting item the group is automatically expanded. User can force expansion by explcitly clicking expand button, groups expanded that way will stay expanded unless user collapses them. Otherwise non-active group will be collapsed automatically.

This PR also changes default size of the dialog to 1000x900 [... cut]