FreeCAD/FreeCAD #11011: [Problem] Clone of a clone does not inherit the colours

scott-pett, 2023-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Cloning a body, correctly inherits the body's colour. But take a clone of that clone and the colour is not inherited.

### Full version info

```shell
[code]
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-12:
Please provide a step-by-step example of how to reproduce the bug

scott-pett, 2023-10-12:
Steps to Reproduce:-

1. Create new document
2. Create Body01
3. Create sketch; draw shape; fully constrain
4. Pad shape
5. Set Body01's appearance so that it's shape colour is something other than default
6. Clone Body01 to create Body001
7. Transform Body001 away from Body01
8. Body001's appearance has inherited the colour of Body01
9. For good measure Create another clone of Body01 to create Body002
10. Transform away and see that it has inherited the original colour.
11. Now clone Body001 or Body002 to create Body003
12. Transform away.
13. It will not have inherited the colours from the clone and will show up as grey.

So it appears that first generation clones will inherit the colour of the parent, but second generation clones do not.

[image]

Roy-043, 2023-10-13:
Confirmed. The issue is related to the fact that the Body and its tip feature can each have their own ShapeColor. The colors of the features of a Body are hidden by default.

In the example Body001 does not inherit the ShapeColor, it is its Clone feature that does.

If this Clone feature is selected for the next step, Body003's Clone feature will inherit the ShapeColor as well.

[code]
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34594 (Git)
Build type: Release
Branch: master
Hash: e9187412d9b3b372ee4c1d7e2fc7a0bf08a2e647
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
[/code]

ElementW, 2026-09-14:
It does *not* correctly inherit the color. On either 1.1.1, 1.1.3, or `FreeCAD_weekly-2026.09.09-Linux-x86_64.AppImage` for that matter.

https://github.com/user-attachments/assets/c475eb57-1064-4397-8fb1-58d0b4bb769f

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/f01619612920a7dd52f1f1b1ef3934716d9ebbe7

The clone Body gets the appearance too, so cloning that Body again keeps the colors.