FreeCAD/FreeCAD #11013: [Overlay] 'Auto Task' doesn't seem to work.

PaddleStroke, 2023-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following #7888 PR merge:
The overlay views have a setting to set either : 

- none
- auto hide
- view on edit
- view on not edit
- auto task

The auto task doesn't seem to work on my end.
From my understanding it is intended for Task view, to show it when there is a taskbox, then hide it when there's nothing.
But on my end nothing is happening.

Also it seems to be redundant with 'View on edit', which does work. Though it doesn't close directly after clicking OK or Cancel, but disappears when it loose focus after clicking the buttons.

Other people seems to report that the option works for them. So perhaps its only on windows.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34642 +15 (Git)
Build type: Release
Branch: asm_card9
Hash: 9cf8885ec1ab18b8d7fdc838ac0b8180ebb7bb21
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * FreeCAD - Shortcut.lnk
  * FreeCAD_FastenersWB-master 0.4.56
  * Ondsel 2023.8.23.02
  * OpenDark 2023.10.6
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-10-12:
@realthunder ping

adrianinsaval, 2023-10-13:
It seems to work fine for me, can show what's your setup and where are you setting the auto task option?

PaddleStroke, 2023-10-13:
I set it here in the heade icons : 
[image]

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34656 +18 (Git)
Build type: Release
Branch: asm_card9
Hash: 325f858fc3d09ed7df86c231abb80ccad6462a99
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * FreeCAD - Shortcut.lnk
  * FreeCAD_FastenersWB-master 0.4.56
  * Ondsel 2023.8.23.02
  * OpenDark 2023.10.6
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
[/code]

adrianinsaval, 2023-10-13:
Keep in mind that the stuff that shows up on Part Design are considered as a task by this, so it only "hides" when the task view is completely empty.

From my observation it seems the "hiding" term here doesn't mean the same as on "Auto hide" mode, the overlayed panel doesn't minimize to the side but instead if there are other panels stacked it switches back to the other when the task view is empty. I don't know if this is intended behavior but certainly the wording of the tooltip could be better.

https://github.com/FreeCAD/FreeCAD/assets/36372335/c934dde5-b954-41fb-bdcd-3ce71365fc1a

PaddleStroke, 2025-01-28:
Coming back to this, it appears you are correct @adrianinsaval . I expected this 'auto task' mode to hide the view when it was empty. But instead it only reduce its height so that other task can be better seen.

Then I guess my issue is a feature request. I want a mode that hides the taskbar when it's empty. Not based on edit mode status.
The reason is that : 
- In PD the taskbox is not empty when not in edit
- In assembly the taskbox is empty when in edit.

luzpaz, 2025-01-28:
> Then I guess my issue is a feature request. I want a mode that hides the taskbar when it's empty. Not based on edit mode status.

Maybe you can rename ticket for clarity?