FreeCAD/FreeCAD #11015: [Overlay] 'Auto Mouse Wheel pass through' doesn't work

PaddleStroke, 2023-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On windows.

'Auto Mouse Wheel pass through' doesn't work.

The mouse clicks are passing through, but the wheel event doesn't seem to pass through.

@realthunder 

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34642 +15 (Git)
Build type: Release
Branch: asm_card9
Hash: 9cf8885ec1ab18b8d7fdc838ac0b8180ebb7bb21
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * FreeCAD - Shortcut.lnk
  * FreeCAD_FastenersWB-master 0.4.56
  * Ondsel 2023.8.23.02
  * OpenDark 2023.10.6
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2023-10-12:
I could try to fix that and refactor this part of code as it was mentioned in the CR if I remember correctly.

adrianinsaval, 2023-10-13:
I can confirm this too. I tried playing around with the delay value for mouse wheel pass through but didn't find anything that worked

realthunder, 2023-10-14:
Probably broken by [this](https://github.com/realthunder/FreeCAD/commit/bdf7af1380c932371f4d01dab243c832dd2c68ec). My Windows build machine is broken at the moment. Please revert this commit, and give it a try.

kadet1090, 2023-10-14:
Yup, it indeed fixes the mentioned issue. I reverted only part of this commit as it was enough to fix the issue, I don't want to break other stuff by accident and change that I reverted is isolated to wheel events.

adrianinsaval, 2023-10-16:
The issue has been mostly solved but there is a remaining problem with techdraw

WandererFan, 2023-10-16:
a) what is [Overlay]?
b) TD processes wheel events in the NavStyle implementation.  Maybe TD is eating an event before Overlay gets it?

WandererFan, 2023-10-17:
void QGVPage::wheelEvent(QWheelEvent* event)
{
    m_navStyle->handleWheelEvent(event);
    event->accept();  // <<<< if somebody else needs this event, they will not get it.
}

adrianinsaval, 2023-10-17:
> a) what is [Overlay]? 

see https://github.com/FreeCAD/FreeCAD/pull/7888

kadet1090, 2023-10-17:
> void QGVPage::wheelEvent(QWheelEvent* event) { m_navStyle->handleWheelEvent(event); event->accept(); // <<<< if somebody else needs this event, they will not get it. }

Hmm, I placed debug log on this line and it never got executed, it does not even go into it. The address of `QGVPage` is also correct (logged from `wheelEvent` that is handled correctly and before `sendEvent`). For some reason, the widget is not processing event sent by the 
```
        QApplication::sendEvent(lastIntercept, &wheelEvent);
```

What is interesting is that if we target specifically `scrollArea->viewport()` it works fine for TechDraw and breaks for the 3D View.

#11077 targets the 3D view specifically and fixes the issue, but I don't feel that this is good solution, feels hacky to me.

PR #11042 (2023-10-16): Gui: Target proper widget when passing wheel event
As mentioned by @realthunder in https://github.com/FreeCAD/FreeCAD/issues/11015#issuecomment-1762922058 the https://github.com/realthunder/FreeCAD/commit/bdf7af1380c932371f4d01dab243c832dd2c68ec commit broke wheel event passing in overlay widgets. Reverting one block fixes the issue, I did not revert the other change as this function is used also in other places which seems to work fine.

Fixes: #11015
commit de030e8f (2023-10-16): Gui: Target proper widget when passing wheel event

Fixes: #11015
PR #11077 (2023-10-18): Gui: Target 3DViewer directly for overlay wheel events
I've no idea why, but for some reason if synthesized event is passed into `QGraphicsView` it is not processed. It however works ok if we pass `graphicsView->viewport()` as receiver of the event. However as `View3DInventorViewer` also indirectly derives from `QAbstractScrollArea` it is also affected by the `qobject_cast<QAbstractScrollArea*>(parent)`, and in that case passing events to viewport does not work. Maybe some Qt experts know why, I'm not one of them.

Changing `qobject_cast` to more specific class seems to work fine for both case [... cut]
commit e8c1a13b (2023-10-18): Gui: Target 3DViewer directly for overlay wheel events

Fixes: #11015
closed 2023-10-18 by wwmayer