FreeCAD/FreeCAD #11016: [Problem] Taskview could resize automatically to minimum size

PaddleStroke, 2023-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The taskview is currently resizing their width to avoid cropping the taskboxes. Which is good.

However they do not shrink back.
So if you open a taskbox that is super wide (partDesign Hole tool for example). Then you have a huge taskview. And if you go to sketcher it looks like this : 
[image]
So the user have to manually shrink the view back down.
Suggestion : Taskbox resize the width automatically to the current taskboxes.

@wwmayer ping as we discussed this before.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34642 +15 (Git)
Build type: Release
Branch: asm_card9
Hash: 9cf8885ec1ab18b8d7fdc838ac0b8180ebb7bb21
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * FreeCAD - Shortcut.lnk
  * FreeCAD_FastenersWB-master 0.4.56
  * Ondsel 2023.8.23.02
  * OpenDark 2023.10.6
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-10-12:
> However they do not shrink back.

The minimum size hint makes the task dialog as big as needed that it's fully visible. However, there are cases where it's hardly possible to interact with a task dialog when it gets resized to its minimum size.

So, if implementing a function to reduce the width then there should be an option to switch this off.

PaddleStroke, 2023-10-13:
> However, there are cases where it's hardly possible to interact with a task dialog when it gets resized to its minimum size.

Can you give examples of such taskboxes ?
I think the correct way to solve this is to :
- Fix those taskboxes so that they offer correct minimum sizes 
- Implement auto-resize to minimum size without parameter.

wwmayer, 2023-10-13:
Placement dialog.

PaddleStroke, 2023-10-13:
Well won't you say that this is a minimum-size-hint issue for this specific taskbox ?
[image]

wwmayer, 2023-10-13:
Should the task view also shrink when you close a dialog? Or should it only shrink to its minimum width when opening a new dialog?

PaddleStroke, 2023-10-13:
I was thinking about when you open a new dialog.
But ideally the task view could hide itself whenever there is nothing in it.

marioalexis84, 2023-10-14:
At the other extreme, this is what I get for some task dialogs (for example, implicit function in FEM workbench): half the screen taken up by the dialog on a 1366X768 monitor.
Maybe not just a size policy in the dialog but also some settings in the spin box as there is too much white space in each box.
Is there a general way to solve this problem or does it need to be solved for each element?
[image]

wwmayer, 2023-10-14:
> Is there a general way to solve this problem or does it need to be solved for each element?

That's probably caused by the minimumSizeHint() implementation of the QuantitySpinBox and the handling of the expression of its base class. If no expression is bound the width of the size hint could be a bit smaller.

adrianinsaval, 2023-10-16:
that particular panel could probably be better organized, vertically there's a lot of space available so it might be better but x, y and z in different rows.

marioalexis84, 2023-10-16:
True. And without repeat the string "Center" with each box, similar as is done in Blender in the Transform fields.

adrianinsaval, 2023-10-16:
Can you tell me how do I reach that panel from the GUI? I do not use FEM frequently, are there other panels there that present that problem?

adrianinsaval, 2023-10-16:
made a PR improving some FEM task dialogs, if there are others with the same problem please mention them in the PR and I will also change them

PaddleStroke, 2023-11-28:
@wwmayer made a commit in his repo ([db1e934](https://github.com/wwmayer/FreeCAD/commit/db1e93435f43295e49d7eea4815dc3791020af3e)) but this has not merged yet.

Regarding the commit, this introduces a parameter which is false by default. Maybe it would be best to enable it by default? Else it would need to be put in the fine tuning wiki page.

wwmayer, 2023-11-29:
https://github.com/FreeCAD/FreeCAD/pull/11027#issuecomment-1764999115

commit 6f76ccfb (2024-05-20): Gui: Restore previous width when closing task dialog

This fixes #11016
closed 2024-05-20 by chennes