FreeCAD/FreeCAD #11017: [Problem] Two direction chamfer with multiple edges produces opposite chamfer on each edge

macdroid53, 2023-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The forum discussion with explanation, images and video:
[https://forum.freecad.org/viewtopic.php?p=712485#p712485](url)

Using to direction chamfer on two edges applies the wide/narrow chamfer on different planes on each edge.

### Full version info

```shell
Found origianly in Linkstage and reproduced in Manjaro and Debian with different builds:

[code]
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34646 (Git)
Build type: Release
Branch: main
Hash: a0af13ad2cca928a7e93ccd8898b5d2751f4fcd5
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.2
  * ZolkoBase 0.0.1
  * Pyramids-and-Polyhedrons
  * lattice2 1.0.0
  * Curves 0.6.13
  * Nodes 0.1.36
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.0
  * Assembly
  * Assembly4 0.50.4
  * freecad.gears 1.0.0
  * sheetmetal 0.3.1
  * Silk 0.1.5
  * kicadStepUpMod 10.19.5
  * A2plus 0.4.60n
  * DynamicData 2.53.0
  * workfeature-macro
  * fasteners 0.4.66
  * ExtremeProDark 2.9.0
  * Defeaturing 1.2.1
  * CurvedShapes 1.0.4
[/code]

[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34594 (Git)
Build type: None
Branch: master
Hash: e9187412d9b3b372ee4c1d7e2fc7a0bf08a2e647
Python 3.11.5, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.53.0
  * Help 1.0.3
  * Defeaturing 1.2.0
  * Curves 0.6.13
  * Silk 0.1.5
  * CurvedShapes 1.0.4
  * Assembly
  * Assembly4 0.50.2
  * fasteners 0.4.66
  * kicadStepUpMod 10.19.5
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.0
  * lattice2 1.0.0
  * IconThemes
  * freecad.gears 1.0.0
  * A2plus 0.4.60n
  * sheetmetal 0.3.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

astraujums, 2024-08-04:
Here is a likely example of this bug.

There are two 7-edge chamfers using two distances, one on either side of an octagonal prism. The first one of them ("Chamfer") works as expected, applying the same one distance to the side shared by all seven edges. The other chamfer ("Chamfer 1"), however, applies one distance to six edges but the other distance to one edge.

This is unexpected.

The expected behavior is that the same distance is applied to the shared face.

This is the unexpected chamfer:
[image]

[image]

This is the expected chamfer:
[image]

This is the model:
[TeltsistabasPoga.zip](https://github.com/user-attachments/files/16488436/TeltsistabasPoga.zip)

Version info:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Latvian/Latvia (lv_LV)

dzid26, 2025-01-07:
Another example with a compound body. - Table on two legs:
[image]
 Top of the table is chamfered, but one edge comes out different.
[CompundBodyTwoDistanceChamferIssue.zip](https://github.com/user-attachments/files/18326746/CompundBodyTwoDistanceChamferIssue.zip)

Apollo-Roboto, 2025-07-01:
I'm only getting started and still getting familiar with the software, but here I am. Doesn't seem like I'm doing anything wrong, at least I can't find a way to flip the direction of this specific edge. I'll add my file too if it helps.

[image]

[chamfer_bug.zip](https://github.com/user-attachments/files/21006245/chamfer_bug.zip)