FreeCAD/FreeCAD #11020: [Problem] Overlay mode "Check Navigation Cube" feature doesn't get correct navi cube position

adrianinsaval, 2023-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Enabling Preferences->Display->UI->Check Navigation Cube doesn't work correctly, it seems to always interpret as if the cube was in the lower right corner, shortening widgets on that side while leaving widgets on the correct side unaffected. Might be due to changes in the NaviCube after @realthunder 's initial implementation of this feature.
[image]

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34651 (Git) AppImage
Build type: Release
Branch: main
Hash: aa5bea4e2e8089bf00f61075a0860dacda42dc8b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
  * fasteners 0.4.66
  * CfdOF 1.21.7
  * Curves 0.6.13
  * Defeaturing 1.2.1
  * DynamicData 2.53.0
  * Dracula 0.0.5
  * SelectorToolbar
  * FC_SU
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * sheetmetal 0.3.1
  * AirPlaneDesign
  * Glass (Disabled)
  * Assembly3 0.12.0
  * OpenDark 2023.10.9
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-15:
@Rexbas is it possible for you to take a look at this ?

Rexbas, 2023-10-15:
Yes, I will take a look.

Rexbas, 2023-10-15:
It appears that the wrong parameter group path was used for "CornerNaviCube". It was "BaseApp/Preferences/View" but should be "BaseApp/Preferences/NaviCube". I made a PR with the fix.

PR #11051 (2023-10-23): Gui: Remove CornerNaviCube from OverlayParams and use directly in OverlayManager
Fixes #11020.

In the file I changed, OverlayParams.cpp, I see a bunch of `// Auto generated code (Tools/params_utils.py:xxx)`. Not sure how that auto generation works but this makes me wonder if the manual changes may get overwritten again if the code is regenerated for some reason.
closed 2023-10-23 by adrianinsaval