FreeCAD/FreeCAD #11023: [Overlay] Overlay stylesheet should be set by preference pack/theme.

MisterMakerNL, 2023-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am not sure if it is broken or it is not implemented, but I cannot set the overlay stylesheet with the preference packs.
Which would be recommended as the preference pack sets the background color and some Qsint colors have overlap. 
Tried adding: 
`        <FCParamGroup Name="MainWindow">
          <FCText Name="OverlayStyleSheet">Dark.qss</FCText>
        </FCParamGroup>`

### Full version info

```shell
Latest dev
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2023-10-13:
from my user config file it looks like the user.cfg line we need to add looks more like this compared to what you have above: <FCText Name="OverlayActiveStyleSheet">OpenDark_Overlay.qss</FCText>

however, I cannot get my overlay stylesheet to even register in freecad at all. I've tried a few different approaches in my package.xml to register the file but none seem to work.

obelisk79, 2023-10-13:
@chennes is something that would fall under your purview?

chennes, 2023-10-14:
Yes. The way it is *supposed* to work as that any QSS file in your mod's directory is found (you don't need to set anything in package.xml). It sounds like that isn't working, so I will have to look into it.

PR #11039 (2023-10-14): Gui: Register overlay search path from preference packs
This PR adds `"<mod>/overlay"` path to search paths of "overlay" prefix. This allows preference packs to register custom overlay stylesheets.

I think that we should think about better ways of doing that kind of stuff. Package metadata could include elements like 
```
<searchpath prefix="qss" path="." />
<searchpath prefix="overlay" path="./overlay" />
```
which should enable package developers to have better control over the search paths. This should however be done in another request and issue.

Fixes: #11023
commit 147cd7cc (2023-10-14): Gui: Register overlay search path from preference packs

This commit adds "<mod>/overlay" path to search paths of "overlay"
prefix. This allows preference packs to register custom overlay
stylesheets.

Fixes: #11023
closed 2023-10-14 by chennes