FreeCAD/FreeCAD #11032: [Overlay] Flickering when changing to PartDesign workbench

wwmayer, 2023-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Every time when changing to the PartDesign workbench there is some flickering of the Task view. For a short moment the view is full of task watchers that disappear again.

This issue is caused by the function TaskView::updateWatcher() that starts a timer to delay the update by 200 ms. When doing the update immediately instead then the flickering disappears.

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34656 (Git)
Build type: debug
Branch: main
Hash: e859961e8ffcee2654b304d0bf55b92be3b64545
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Core

### Anything else?

The problem is caused by bf9584a90ba9

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-17:
> This issue is caused by the function TaskView::updateWatcher() that starts a timer to delay the update by 200 ms. When doing the update immediately instead then the flickering disappears.

Who wants to fix this ?

luzpaz, 2023-10-24:
Per wmayer the diagnosis:

> This issue is caused by the function TaskView::updateWatcher() that starts a timer to delay the update by 200 ms. When doing the update immediately instead then the flickering disappears.

bgbsww, 2023-10-31:
I can fix this if you assign me.  Note that this appears to work fine if you rip out the timer and just replace the contents of TaskView::UpdateWatcher with the contents of TaskView::onUpdateWatcher.  I can't see any reason for the timer to exist.

luzpaz, 2023-10-31:
Go for it @bgbsww

PR #11249 (2023-11-02): Fix #11032 flickering when switching to Part Design
Fixes #11032
commit 8baf75af (2023-11-02): Fix #11032 flickering when switching to Part Design

* Remove unneeded timer on taskview updates to fix #11032

* Remove obsoleted class member
closed 2023-11-02 by wwmayer