FreeCAD/FreeCAD #11036: [Problem] Image: size of ImagePlane changes when copied in a different document

alafr, 2023-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I copy an ImagePlane object from a document to another document, the XSize and YSize of the new copy are different than the original (in the example the copy is 359x466 mm, the original is 133x172 m).
To reproduce:

-   open attached document
-   copy the image plane and the rectangle that has the same size (Ctrl+C)
-   create new document
-   in the new document, paste (Ctrl+V)
-   the image is now smaller, needs a lot of zoom to see it

[image]
[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * BIM
  * Defeaturing 1.2.1
[/code]
```

### Subproject(s) affected?

File formats

### Anything else?

[imageplane_test.zip](https://github.com/FreeCAD/FreeCAD/files/12905351/imageplane_test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-10-14:
Confirmed. However, when adding the copy to the source document then the size is correct.

alafr, 2023-10-14:
It's exact, I scaled the image to match real world dimensions (I don't remember if I used the calibrate button or I manually set the XSize & YSize, but in both cases the behaviour is the same) and the copied image has its size reset to the size it had when it was first inserted.
I also confirm that there is no issue when the copy is in the same document.
The issue happens with Copy / Paste in different documents, drag / drop in different documents, Merge Project command.
There is no issue with Link.

sushanthvutukuri, 2024-01-29:
Hello, I was wondering if I could work on this issue?

PR #12282 (2024-02-12): Avoid overriding ImagePlane size during import
When copying&pasting an ImagePlane whose ImageFile property is a PropertyFile we'd get two updateData notifications, one when the property value is restored (which doesn't do much as the file doesn't exist yet) and another when the RestoreDocFile method is invoked.

The solution is simple, do not set Xsize and Ysize when importing the object as we expect the properties to already have the correct values.

Closes #11036
commit 5af79334 (2024-02-12): Avoid overriding ImagePlane size during import

When copying&pasting an ImagePlane whose ImageFile property is a
PropertyFile we'd get two updateData notifications, one when the
property value is restored (which doesn't do much as the file doesn't
exist yet) and another when the RestoreDocFile method is invoked.

The solution is simple, do not set Xsize and Ysize when importing the
object as we expect the properties to already have the correct values.

Closes #11036
commit 2e025dfc (2024-03-13): Avoid overriding ImagePlane size during import

When copying&pasting an ImagePlane whose ImageFile property is a
PropertyFile we'd get two updateData notifications, one when the
property value is restored (which doesn't do much as the file doesn't
exist yet) and another when the RestoreDocFile method is invoked.

The solution is simple, do not set Xsize and Ysize when importing the
object as we expect the properties to already have the correct values.

Closes #11036
closed 2024-02-12 by chennes