FreeCAD/FreeCAD #11043: [Problem] Preference Packs have no control over QT Search Paths

kadet1090, 2023-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some places in FreeCAD relies on `QDir` search path feature (https://doc.qt.io/qt-6/qdir.html#setSearchPaths). Currently it is done in the constructor of `Gui::PreferencePack` but added paths are hardcoded and preference pack authors have no control over them. Search paths associated with preference pack should be defined in the metadata file, solution must be backwards compatible with older packages that do not specify that values explicitly.

As mentioned in https://github.com/FreeCAD/FreeCAD/pull/11039 we can do something like:
```
<searchpath prefix="qss" path="." />
<searchpath prefix="overlay" path="./overlay" />
```

Where `.` means the root of preference pack. To keep backwards compatibility we can specify implicit defaults:
```
qss: .
css: .
overlay: ./overlay
```

That can be changed using explicit `<searchpath />` declaration. We should also allow declaring more than one `<searchpath />` for given prefix. I don't know if there is some valid use case for that, but I don't see any reason to forbid that either. 

### Full version info

```shell
0.22 dev
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-10-14:
Seems totally reasonable to me.

RohanMishra315, 2023-12-15:
Hey @luzpaz could you help me to solve this !

kadet1090, 2023-12-16:
@RohanMishra315 If you want to contribute changes I can help you a bit with this feature on Discord: https://discord.gg/GbAfRTWF #development

luzpaz, 2023-12-22:
@RohanMishra315 should I assign this ticket to you ?

RohanMishra315, 2023-12-23:
Yes , please.