FreeCAD/FreeCAD #11045: [Problem] Freecad fails to start after uninstalling disabled addon

adrianinsaval, 2023-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Up until recently I was using my fork of the Glass addon, when the panels where refactored I disabled this addon as it wasn't working properly and now that the overlay PR has been merged I had no more use for it, so I went to the addon manager and uninstalled this addon, then I wanted to change something in the preference page but had this error in report view when trying to open it:

> 17:47:11  C++ exception thrown for 'Gui::Dialog::DlgSettingsGeneral' (boost::filesystem::status: Permission denied [system:13]: "/home/adrian/.local/share/FreeCAD/Mod/Glass/package.xml")

and IIRC the preference page did not open.

Now I tried opening freecad and it can't even start, I get this error in terminal:

> Application unexpectedly terminated: boost::filesystem::status: Permission denied [system:13]: "/home/adrian/.local/share/FreeCAD/Mod/Glass/package.xml"

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34662 (Git)
Build type: None
Branch: main
Hash: 310c052134cf728249cc24c8af3db657e9289c6d
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-10-14:
@wwmayer @PaddleStroke @chennes

wwmayer, 2023-10-14:
Does this file exist? /home/adrian/.local/share/FreeCAD/Mod/Glass/package.xml

chennes, 2023-10-14:
Uninstalling *should* have removed the Mod/Glass subdirectory -- out of curiosity, did Glass provide a preferences page? Of course, regardless, on Linux there should have been no problem removing the directory, and the Addon Manager didn't give an error, so I really hope that it did indeed remove that path! `package.xml` should never be *required*, except for a Preference Pack -- if it doesn't exist for a normal mod it should just be silently ignored. Maybe it didn't get deleted, but somehow ended up corrupted?

adrianinsaval, 2023-10-16:
so this is way more niche than I initially thought but it still a problem that needs to be adressed IMO (specially that freecad should not fail to launch do not being able to access a mod directory!).

For context I was running my own fork of the Glass addon, I have a directory on my home where I clone all the repos I work on so I had cloned Glass there and symlinked that into the FreeCAD/Mod directory, when uninstalling the addon instead of removing this link or any of the directory contents the original directory ended up with these permissions:
[image]

restoring read permissions on it "fixes" the problem, and the issue is reproducible when trying to uninstall again the addon. One correction though, the preference page does launch anyways with that error but freecad itself doesn't

wwmayer, 2023-10-16:
The function to blame is PreferencePackManager::FindPreferencePacksInPackage. It has this check `fs::exists(packageMetadataFile) && fs::is_regular_file(packageMetadataFile)` that raises the exception but it's outside the try/catch(...) block and thus the exception is thrown up to the main function which the exits the application.

closed 2023-10-16 by chennes

adrianinsaval, 2023-10-16:
thanks taht fixed the problem with freecad not starting, @chennes do you think we should fix the problem with the addon manager not handling symlinks properly at uninstall?

chennes, 2023-10-16:
If the fix is easy I see no downside. I don't remember if I was doing anything more complicated than telling Python to remove the directory.

commit 15470035 (2023-10-16): Gui: fixes #11045: Freecad fails to start after uninstalling disabled addon

The PR refactors FindPreferencePacksInPackage() and splits program logic from exception handling. It also handles std::exception to give more
information about a possible failure
commit 916a1146 (2023-11-09): Gui: fixes #11045: Freecad fails to start after uninstalling disabled addon

The PR refactors FindPreferencePacksInPackage() and splits program logic from exception handling. It also handles std::exception to give more
information about a possible failure