FreeCAD/FreeCAD #11046: [Problem] Installing FreeCAD with a custom file directory name

justSomeTester, 2023-10-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Installing FreeCAD 0.21.1 for Windows 10 using the 64-bit installer option, then including the string: " ; ", in the file directory name at any point in the name, successfully installs FreeCAD but FreeCAD will refuse to launch.
Examples: D:\ ; , D:\aa;aa

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Europe (en_150)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-15:
@justSomeTester What's the error ? Can you launch it from the command line and see if it spits out an error ?

Syres916, 2023-10-15:
Surely a semi colon is against Windows file system principles therefore the installer shouldn't even allow the user to input that character let alone continue with the installation. I would have expected NSIS to have this setting by default. Only the path type environment variables would have a semi colon in them as a separator.

justSomeTester, 2023-10-15:
> @justSomeTester What's the error ? Can you launch it from the command line and see if it spits out an error ?

Trying to start FreeCAD via cmd throws the following system errors about the following missing files:
FreeCADGui.dll , FreeCADApp.dll , FreeCADBase.dll , Qt5Widgets.dll .

I did the same with two separate Windows 10 devices and both of them had the same issue. Although of course this shouldn't be a major bug since one can just avoid putting a semi colon into the name. I just figured I would point this out. I otherwise have no issues with the installation of FreeCAD.

luzpaz, 2023-10-31:
@chennes the windows installer logic should theoretically prevent this from occuring. Can I put this ticket on your radar? or @adrianinsaval's ?

adrianinsaval, 2023-10-31:
I'll Google around if this is something we can forbid in the installer

luzpaz, 2023-11-08:
@alexbeer2048 do you have time to take a look at this ?

alexbeer2048, 2023-11-11:
> @alexbeer2048 do you have time to take a look at this ?

I can look into this.

Something up front:
";" is a valid character in a Windows path. I can create a folder with this character.

I think in theory the program should still run. But maybe the way we load our DLLs does not allow it.

bgbsww, 2023-12-10:
So, a simple search on ';' and ";" in the source code reveals multiple places where we appear to be working on a path using this literal.  I can't test on Windows, but it might be a starting place for someone who can.  There are around 130 hits, and I see at least 3 or 4 that are around path strings.  I do mean searching on literally those three character strings.

luzpaz, 2023-12-13:
> I think in theory the program should still run. But maybe the way we load our DLLs does not allow it.

Can you explain this hypothesis ?

maxwxyz, 2024-06-13:
any news on this?

closed 2025-02-18 by github-actions[bot]