FreeCAD/FreeCAD #11052: [Problem] Poor implementation of part container in task suggestions

GentlemanRider, 2023-10-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When starting a new document in part design, the task panel only proposes to create a new body. It would be useful to include also an option to create a new part.

When a part container is selected, there is no task suggestion at all. There should be options again to create part or body.

[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git)
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * BOLTSFC
  * Curves 0.6.5
  * DynamicData
  * ExtremeProLight 1.6.0
  * fasteners
  * fcgear 1.0.0
  * LCInterlocking
  * parts_library
  * ProDarkThemePreferencePack 1.0.0
  * Pyramids-and-Polyhedrons
  * TabBar
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-10-15:
I believe this ticket is due to a misunderstanding of the Part Design vs. Part workbenches. Not an uncommon mistake by any means!

luzpaz, 2023-10-16:
Related to #9748

GentlemanRider, 2023-10-16:
> I believe this ticket is due to a misunderstanding of the Part Design vs. Part workbenches. Not an uncommon mistake by any means!

No it's not.

Let's say I want to model a laser cut box. I want a Part (yellow icon container) called BoxMain with several bodies inside. It all happens in part design and allows me to draw the various box parts in place, and then move (or assembly) the thing as a whole.

In an empty document (as well as inside of a Part) creating a Part makes as much sense as creating a body. Take a look at the attached model tree if it's not clear what I mean.

[image]

GentlemanRider, 2023-10-16:
> Related to #9748

So and so. I agree that the name may be confusing but I think I understand the functionality of parts and bodies in part design. At least, I use them both all the time and it works.