FreeCAD/FreeCAD #11055: [Problem] FEM Results pipeline cannot store the results from multiple frames

FEA-eng, 2023-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If an analysis consists of multiple frames of results (increments in the case of nonlinear simulations or mode shapes in the case of frequency and linear buckling analyses), a separate results pipeline object is created for each frame, making postprocessing inconvenient (and it's not even possible to export all these objects to ParaView at once). Furthermore, multiple CalculiX_Results object are also created, making it even worse:

[image]

The idea is to remove the CalculiX_Results objects (while extending the capabilities of the results pipeline) as described in #10821 and make the pipeline support results from multiple frames, adding an option to select a frame to display. This would also make it possible to create animations of results in the future. ParaView handles vtk groups, displays them as a single object and offers options to change frames so it should be possible in FreeCAD as well.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git)
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2023-10-16:
Agree with unifying and homogenizing results an pipeline objects.

FEA-eng, 2024-02-18:
@maxwxyz Can you remove the Bug label? This one is a feature request.

ickby, 2024-12-20:
A "prove of concept" implementation for review: https://github.com/ickby/FreeCAD_sf_master/tree/FEM_multistep_result
Discussion: https://forum.freecad.org/viewtopic.php?t=93121&sid=30411efd1aade7ea124f0e2b7cf1d466

ickby, 2025-03-23:
Solved with merging #19610

closed 2025-03-23 by maxwxyz