FreeCAD/FreeCAD #11057: [Problem] FEM Examples are unnecessarily duplicated as calculix and ccxtools

FEA-eng, 2023-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FEM Examples grouped by the solver, are unnecessarily duplicated as calculix and ccxtools while this is essentially the same:

[image]

What's more, sometimes it's called "ccxtools" while sometimes it's called "cxxtools":

[image]

https://wiki.freecad.org/FEM_SolverCalculixCxxtools

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git)
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

lyphrowny, 2024-03-10:
How thorough do we need to be in removing "calculix" from the examples? (It's "ccxtools", that goes by the "CalculiX Standard", right?)

Options

1. Just don't list "calculix" in the examples tree
2. From all example files remove "calculix" as a solver
3. Remove "calculix" completety

The 1st is the least invasive, the 3rd - the most. The 3rd, however, has potential in another area: since 0.22 "calculix" should be obsolete ([link](https://wiki.freecad.org/FEM_SolverCalculiX)), so removing it now will resolve 2 issues

After a bit of tinkering with the 3rd option got super confused in what's going on under the hood. So the 2nd option seems the most promisable: fixes this issue and makes a step forward removing the "calculix" - will go with it

While we're on the "ccxtools", "calculix" discussion, maybe it would make sense to rename "ccxtools" to "calculix"? Not now, but when the "calculix" is be removed

FEA-eng, 2024-03-11:
> After a bit of tinkering with the 3rd option got super confused in what's going on under the hood. So the 2nd option seems the most promisable: fixes this issue and makes a step forward removing the "calculix" - will go with it

Right, it should be sufficient for now. Ultimately, it would be good to get rid of all the references to the removed "new framework" solver but it's not so urgent.

> While we're on the "ccxtools", "calculix" discussion, maybe it would make sense to rename "ccxtools" to "calculix"? Not now, but when the "calculix" is be removed

Sure, that should be done at some point as well. It's way too confusing currently.

lyphrowny, 2024-03-31:
Are we sure, that `calculix` needs to be removed and not `ccxtools`? I cannot draw a definitive conclusion: sometimes `ccxtools` seems more "advanced", compared to `calculix`; sometimes it's the other way around

Cons to `ccxtools`

- Currently located in "unexpected" places

  All of the solvers are located under `femsolver` directory, while `ccxtools` is in `femtools` directory (why is it there? why wasn't it moved? [because there were plans on its removal?])

- Not a part of framework

  All other solvers are a part of a framework - have methods in common. Having a common interface should simpify working with different solvers (in code)

Cons to `calculix`

- `ccxtools` seems to have more code

  This code seems to be responsible for various checks, like ccx version, having any materials, and non-positive jacobians

Both `ccxtools` and `calculix` seem to be just interfaces - they call lots of common methods, which are responsible for writing .inp files and reading result files

So I propose not removing `calculix` because it's unfinished, but finishing it, taking the best from `ccxtools`

What do you think, @FEA-eng? My analysis is quite superficial, so I might've missed the bigger picture

FEA-eng, 2024-03-31:
@lyphrowny First of all, there should be only one CalculiX solver, no need for two. Second, **FEM_SolverCalculiX** a.k.a **Solver CalculiX (new framework)** was removed from GUI because it was unfinished and didn't offer anything else than the other solver (**FEM SolverCalculixCxxtools** a.k.a. Solver CalculiX Standard) which was the recommended one (due to negative jacobian checks and so on). Some explanation here: https://github.com/FreeCAD/FreeCAD/issues/10823#issuecomment-1736019693

PR #12876 (2024-04-01): FEM: Fix cxx typo and remove calculix from solvers tab in examples
closes #11057 

- "calculix" was removed from the "solvers" tab in examples
[image]
- "cxx" typo was fixed in GUI, in a class name
[image]
- convert every "FEM_SolverCalculixCcxtools" and its variations to CamelCase: "FEM_SolverCalculiXCcxTools"

Needs recompilation to work

Compiled and tested on WIndows
`run FEM TestSolverCalculix tests` shows `Unknown or unsupported solver type: calculix. No solver object was created.`, which is expected, see https://github.com/FreeCAD/FreeCAD/blob/9099d7588491aed417a2796ec5cfa7cd98a48a88/sr [... cut]
closed 2024-04-01 by chennes