FreeCAD/FreeCAD #11058: [Problem] There's no information about the solvers supported by each FEM analysis feature

FEA-eng, 2023-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some analysis features in FEM are supported by multiple solvers while some are supported only by selected solvers. However, there's no information at all about this, making it highly confusing for new users. Here's what we can do about it:
1. Add this information on the wiki - that's easiest to do but not sufficient, in my opinion
2. Add this information to the tooltips in GUI
3. Disable (gray-out) the analysis features not supported by the currently selected solver - I would choose this approach, eliminating the risk of mistakes like this: https://forum.freecad.org/viewtopic.php?t=81964

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git)
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2023-10-17:
in _3._, the restriction should be in both directions since you can add the constraints first and then the solver.

FEA-eng, 2024-07-16:
1 is done - I added this info on the wiki (using a new field in the _GuiCommand_ table). Now the last option doesn't seem to be a good idea as long as multiple solvers can be added to the same analysis container and it's so easy to switch solvers. Thus, I added another proposal - showing warnings/errors when submitting analyses using features not supported by the current solver. It might be the best option (apart from the tooltips) since the analysis shouldn't run 100% normally with unsupported features.

FEA-eng, 2025-01-30:
Still relevant