FreeCAD/FreeCAD #11060: Statusbar selection of Project unit system does not work

Roy-043, 2023-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While trying out the new Project unit system I notice a glitch.

* Delete/rename your cfg file.
* Start FreeCAD.
* Create a new document.
* In the statusbar select: "Building Euro".
* switch to Part.
* Create a Box.
* Check the dimensions in the Tree view.
* Length: 1,00 cm (OK).
* Open the [wiki]Preferences Editor[/wiki].
* "Unit system" is "Building Euro".
* "Project unit system" is "Standard" (Not correct).
* Change the Project unit system to "Imperial decimal"
* Result: the statusbar and the Tree view show inches.
* In the statusbar select: "Metric small parts".
* Open the Preferences Editor.
* "Unit system" is "Metric small parts".
* "Project unit system" is "Imperial decimal" (Not correct).

Expected behavior: the statusbar selector should only be linked to the Project unit system.

Related to #9521.

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git)
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-10-17:
> While trying out the new Project unit system I notice a glitch.

It's not really a bug because it was never implemented. The current workflow is a bit awkward because you have to change this in the preferences after having created a new document. It would be more logical to move this setting to the _Project Information_  dialog and for quick access to the status bar.

bgbsww, 2023-11-02:
I'm playing with some code here, and I've realized that the goal isn't perfectly clear, so here's a proposal:

-  A Gui field in _Project Information_  with a pulldown to specify the unit system schema.
- Value of this field is stored with the document,
- The Status Bar Dimension widget at the bottom of the main window both reflects and updates this setting.
- When a new document is created, this value is initialized with the Preferences -> General -> Unit System and decimals.
- The Preferences -> General -> Project Unit System and ignore options are removed.

I'm hand waving a bit over whether there should be a decimals option in the _Project Information_ as well.  Arguments either way would be helpful.

Does this make sense, or is there a deeper use of the ignore option that I'm missing?

wwmayer, 2023-11-02:
> Does this make sense

IMO, yes it does.

bgbsww, 2023-11-03:
I'm happy to be assigned here, and resolve any concerns after folks check out the PR.

Roy-043, 2023-11-04:
IMO an ignore option should be available in the preferences.

bgbsww, 2023-11-04:
Where would that button go, and exactly would the ignore button do, @Roy-043 ?  As I think through the workflows, I don't run into a need, but I'm happy to add one if there's a description of the workflows desired.

When you create a new file, it copies your preference settings.
If you don't like that unit system for this file, you change it either in the Project Info or on the Statusbar.

If you open an existing file and you don't like the unit system, you change it.

I can see scenarios of either (A) wanting to save a file that automatically opens in the preferred units of the opener ( stored on the document ), or (B) wanting to set your FreeCAD to always open files in your unit system ( stored in the preferences ), but they both seem less desirable to me than the behavior I described and then implemented.

Since you stated "in the preferences", I'm assuming you are looking for some version of (B).  What, and why?

Roy-043, 2023-11-05:
You are not considering 3rd party files. If you work with such files a lot (for example because you assist in the Forum or do bug triage) you would want to be able to ignore the project unit system.

The ignore option would be a checkbox in the prefs (as it is now)

I can also imagine users who do not want to set a project unit sytem. Which may mean another checkbox in the prefs.

bgbsww, 2023-11-05:
Thanks, that was helpful.  See latest version of PR, which in addition to the above:

- Provides an ignore checkbox in the prefs
- Overrides the project unit schema when checked
- Edits to the statusbar dimension widget or the project information can change the project unit schema
- The statusbar dimension widget never changes the default unit schema in the preferences, even when no document is open
- Files are marked as changed after a unit change, regardless of what the state of the ignore checkbox is

Roy-043, 2023-11-09:
I agree with that approach. Thanks for looking into this.

bgbsww, 2023-12-04:
Flagging that #11266 was merged, so I think this can finally be closed?

closed 2023-12-05 by Roy-043