FreeCAD/FreeCAD #11069: Shapestring error

dross012, 2023-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm using the latest stable version, no custom preferences except grid size.
When I try to create a shapestring I get the following error.
Thank you!

22:55:57  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\drafttaskpanels\task_shapestring.py", line 148, in accept
    self.createObject()
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\drafttaskpanels\task_shapestring.py", line 169, in createObject
    x = App.Units.Quantity(self.form.sbX.text()).Value
<class 'ValueError'>: syntax error

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-10-17:
You need to provide a step-by-step scenario. Including which coordinate values you have entered. With the current information I can't reproduce.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

dross012, 2023-10-18:
I redesigned the project and no longer need Shapestring. Thank you for your reply.

luzpaz, 2023-10-18:
@dross012 if you found a bug..help us reproduce it

dross012, 2023-10-18:
Attached is the things I did to get the error message. I also attached the project I'm working on. I hope this helps you. If there is anything else you need, please let me know. Thank you for your help!!!

luzpaz, 2023-10-19:
@dross012 nothing attached. You need to drag drop the file in zipped format in to github text box for it to upload.

dross012, 2023-10-19:
[Shop.zip](https://github.com/FreeCAD/FreeCAD/files/13037403/Shop.zip)

yorikvanhavre, 2023-10-19:
It's strange... You are working in imperial units, right?
I can reproduce the problem once, then switching to centimeters works, then switching back to imperial works too

dross012, 2023-10-19:
Yes, I'm using feet and inches.

dross012, 2023-10-20:
Update: I was using Build US not Imperial

luzpaz, 2023-11-18:
@Roy-043 can you take a stab at reproducing ?

Roy-043, 2023-11-20:
There is an issue with the Building US unit system. This may be related.
See #11345.

Roy-043, 2023-11-21:
This is indeed related to the Building US unit system.

Roy-043, 2023-11-22:
Closing this as Duplicate.

closed 2023-11-22 by Roy-043