FreeCAD/FreeCAD #11071: [Problem] macOS signing needs entitlements for addons that ship binary libs to work

kroko, 2023-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**TL;DR**

FreeCAD for macOS has to be signed with [entitlements](https://developer.apple.com/documentation/bundleresources/entitlements) so that any FreeCAD addons that ship with loadable binaries can be used.

***

A good way to replicate the issue is using latest stable macOS package [FreeCAD-0.21.1](https://github.com/FreeCAD/FreeCAD/releases/tag/0.21.1) and trying to install [Assembly 3](https://github.com/realthunder/FreeCAD_assembly3). Via addon manager, or _manually_, does not matter.

The issue is observed on on

- macOS Sonoma 14.0 (arm64)
- macOS Monterey 12.7 (Intel)

It does not has to do with architecture, but macOS security enforcements per se.

To reproduce

Check that FreeCAD is signed
```sh
codesign --display --verbose /Applications/FreeCAD-0.21.1/FreeCAD.app
```

Check that FreeCAD is does not have any entitlements attached
```sh
codesign -d --entitlements - /Applications/FreeCAD-0.21.1/FreeCAD.app
```

Install Assembly 3. Loading of the addon will fail, console spits

```
<asm3.main> init_gui.py(14): no solver backend found
```

Listing modules in console finds `py_slvs`

```py
help("modules")
```

`py_slvs` is in the path

```py
print(sys.path)
```

Either location, depends on how `py_slvs` was installed
```
~/Library/Application Support/FreeCAD/AdditionalPythonPackages
~/.local/lib/
```

Using console directly to invoke import shows the issue

```py
from py_slvs import slvs
```

<details>
  <summary>Traceback</summary>

```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Applications/FreeCAD-0.21.1/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/Users/reinis/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/slvs.py", line 10, in <module>
    from . import _slvs
  File "/Applications/FreeCAD-0.21.1/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
ImportError: dlopen(/Users/reinis/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so, 0x0002):
 tried: '/Users/reinis/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so'
 (code signature in <E87CC3D0-649F-36F7-8CA2-9DE20EB366CD> '/Users/reinis/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs),
 '/System/Volumes/Preboot/Cryptexes/OS/Users/reinis/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so' (no such file),
 '/Users/reinis/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so' (code signature in <E87CC3D0-649F-36F7-8CA2-9DE20EB366CD>
 '/Users/reinis/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs)
 ```
</details>

```mapping process and mapped file (non-platform) have different Team IDs``` being the key.

The solution is to have  entitlements plist file, containing [com.apple.security.cs.disable-library-validation](https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_security_cs_disable-library-validation) that is passed for signing

entitlements.plist
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>com.apple.security.cs.disable-library-validation</key>
	<true/>
</dict>
</plist>
```

There are also some other keys to look at, such as

```xml
	<key>com.apple.security.cs.allow-dyld-environment-variables</key>
	<true/>
	<key>com.apple.security.cs.allow-jit</key>
	<true/>
	<key>com.apple.security.cs.allow-unsigned-executable-memory</key>
	<true/>
	<key>com.apple.security.cs.disable-executable-page-protection</key>
	<true/>
```

I have not used Cmake to specify `entitlements`, but it seems it has [variable for `CPACK_BUNDLE_APPLE_ENTITLEMENTS`](https://cmake.org/cmake/help/v3.6/module/CPackBundle.html)

which yealds something like

```cmake
# ...
set_target_properties(
  # ...
  CPACK_BUNDLE_APPLE_ENTITLEMENTS "${CMAKE_CURRENT_SOURCE_DIR}/MacAppBundle/Entitlements/entitlements.plist"
)
#...
```

I did not dig that deeply how package is signed, but it seems, that there is `src/Tools/macos_sign_and_notarize.sh`, that uses `codesign` directly.

One can pass `--entitlements` to `codesign` too, see [this discussion](https://developer.apple.com/forums/thread/129977).

I could do a PR, but IMHO going through such PR may take longer than for the one who is responsible for macOS signing, has all the environment set up already _to go_ to add entitlements to the worflow.

***

BTW, _workaround_ is to disable protections in macOS

```sh
# Disable System Integrity Protection (SIP) via recovery terminal
csrutil disable
# Actually IMHO disabling SIP is enough, but let's theow in SecAssessment system policy security disable too
sudo spctl --global-disable
```

-> Assembly 3 smootly loads after disabling sec in macOS.

### Full version info

```shell
OS: macOS 14.0
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * fasteners 0.4.67
  * Assembly4 0.50.4
  * Assembly3 0.12.0
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kroko, 2023-10-17:
Obvious addition ->  I will happily test _signed w/ entitlement build_ on different macOS versions (10.15.7 x64, 12.7 x64, 14.0 arm64) if such is created as a result of this ticket. 😄

chennes, 2023-10-18:
Can you please see if the version in this DMG works? I only added the entitlements to the outermost layer of the signing. If this doesn't work I'll also add it to the inner executables.

https://plsok-my.sharepoint.com/:u:/g/personal/chennes_pioneerlibrarysystem_org/ER4jFIRDn29FioUJig3CEkABChjUc8UF7Dn6bM8k6lXfAg?e=ne5021

kroko, 2023-10-19:
I can see entitlement

```sh
╰─➤  codesign -d --entitlements - /Applications/FreeCAD-0.21.1/FreeCAD.app
Executable=/Applications/FreeCAD-0.21.1/FreeCAD.app/Contents/MacOS/FreeCAD
[Dict]
	[Key] com.apple.security.cs.disable-library-validation
	[Value]
		[Bool] true
```

I just bit the bullet and renewed apple dev, so i have signed `_slvs.so` at hand, which gives the best way to test this - both signed, both have different IDs.

```sh
╰─➤  codesign --display --verbose ~/Library/Application\ Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so                                         1 ↵
Executable=/Users/reinis/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so
Identifier=_slvs
Format=Mach-O thin (arm64)
CodeDirectory v=20500 size=4721 flags=0x10000(runtime) hashes=142+2 location=embedded
Signature size=8979
Timestamp=19 Oct 2023 at 02:57:21
Info.plist=not bound
TeamIdentifier=J9GU7PBBUS
Runtime Version=12.1.0
Sealed Resources=none
Internal requirements count=1 size=168
```

But unfortunately it does not work, same traceroute as above.

> If this doesn't work I'll also add it to the inner executables.

```
/FreeCAD.app/Contents/Resources/bin/python # my bet is this, as it is python, that tries to load
/FreeCAD.app/Contents/Resources/bin/freecad # there are cases where freecad on macos will launch directly via this bin, not "main bundle binary" at /Applications/FreeCAD-0.21.1/FreeCAD.app/Contents/MacOS/FreeCAD, so yeah
/FreeCAD.app/Contents/Resources/bin/freecadcmd # because..
```

P.S. 

`/FreeCAD.app/Contents/Resources/bin/` also holds `coin-config`, `pip`, `pyside2-rcc` which are not signed.

chennes, 2023-10-22:
OK, I have [another one for you to try](https://plsok-my.sharepoint.com/:u:/g/personal/chennes_pioneerlibrarysystem_org/ER4jFIRDn29FioUJig3CEkABChjUc8UF7Dn6bM8k6lXfAg?e=LpSX8F): I've refactored the signing and it should now be using those entitlements in all cases.

kroko, 2023-10-23:
It works now. ASM3 (signed `_slvs.so`) is loaded on app launch. 🥳 

So it is now up to addon creators to sign binaries, if they ship such (i.e. ASM3 and solvespace).

It seems that the app is not notarized and stapled though.

```sh
╰─➤  spctl -a -vvv -t install /Applications/FreeCAD-0.21.1/FreeCAD.app
/Applications/FreeCAD-0.21.1/FreeCAD.app: a sealed resource is missing or invalid

╰─➤  xcrun stapler validate /Applications/FreeCAD-0.21.1/FreeCAD.app
Processing: /Applications/FreeCAD-0.21.1/FreeCAD.app
FreeCAD.app does not have a ticket stapled to it.
```

PR #11246 (2023-11-06): Tools: Add entitlements to MacOS signing
Fixes #11071
closed 2023-11-06 by chennes