FreeCAD/FreeCAD #11072: [Request] UI improvement: Add properties view to Right click and put it a seperate window.

MisterMakerNL, 2023-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is an old one that never got done.
I was thinking/hoping now @PaddleStroke is doing the transparency in the right click menu, it is an easy thing to do.
The idea is that properties view is something advanced for advance users and should not be necessary for simple things. So should be off by default. A common user should just be able to right click on something and open properties. Just like a lot of programs work.
Also myself I only use it for setting features to refined, and therefore would rather have this in the right click than in a panel.

But how it works now you either have it on or you don't. One could say it needs to use a short-cut, but this complicate things unnecessary.

But I think the request is pretty obvious. 
@Zolko-123  did a pre-study to this:
https://forum.freecad.org/viewtopic.php?t=33831
[image]

### Full version info

```shell
NA
```

### Subproject(s) affected?

None

### Anything else?

UI
UX
Properties

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-10-17:
> The idea is that properties view is something advanced for advance users and should not be necessary for simple things. So should be off by default. A common user should just be able to right click on something and open properties. Just like a lot of programs work.

+1 this is on my mind for some time already.

yorikvanhavre, 2023-10-18:
I love the idea of being able to turn the properties view on/off with a key shortcut. Sometimes one has a very long tree and it helps to navigate it when the properties view is off.
Not sure about the use of F1 though, because ATM F1 is used for help, and that's also pretty common...

luzpaz, 2023-10-24:
Anyone want to take point on this ?

MisterMakerNL, 2023-11-05:
Ok here is the idea how we can probably do this real easy.
We just add a "properties" option to the right click, and it would just pop-up the properties panel.
Right clicking on a object switched the panel to the object anyway.
I already made a [macro to pop-up the panel.](https://github.com/MisterMakerNL/Freecad_Hide_docks_macros/blob/main/show_combo_prop_view.FCMacro)

[We can also make an option to prevent the panel to dock](https://forum.qt.io/topic/85996/how-to-disable-snapping-docking-of-qdockwidget-to-the-centeral-widget/2), so it stays a window. This is very much preferable since if it would pop-up docked the user will probably not notice it.

pierreporte, 2023-12-13:
I just had an idea while thinking about one of my comment: https://github.com/FreeCAD/FreeCAD/issues/11637#issuecomment-1839569787. The properties could be a panel similar to tasks. Currently, when the task panel is hidden, the helper is pretty much invisible and I proposed to make it more obvious, basically as a vertical tab. There could be more than one tab and the properties could be one of them.

This is inspired by Solidworks that has exactly this:

[image]

When the panel is closed, it is very discreet:

[image]

These tabs could be used customized and add-ons could add their own. You can see Solidworks screenshots where the linked PDM has its own tab. [Ondsel Lens](https://github.com/Ondsel-Development/Ondsel-Lens) creates its own panel, as you can see on the README and picture below, so it would benefit from this. There could be some contextual help, a file browser for the assembly and standard parts (like Solidworks).

[image]

@sliptonic do you think this is interesting for Ondsel?

@obelisk79 what do you think in terms of UI/UX?

sliptonic, 2023-12-15:
> @sliptonic do you think this is interesting for Ondsel?

Yes.  Very interesting.  We've been thinking about several different ways to make the Lens PDM functionality blend in to the wallpaper better.  This is certainly an improvement over the current state of affairs

pierreporte, 2023-12-16:
@sliptonic The tabs already exist so it may be some foundations for this feature already. See @realthunder comment here: https://github.com/FreeCAD/FreeCAD/issues/11637#issuecomment-1858703345

obelisk79, 2023-12-16:
In terms of UI/UX I like this approach, particularly if the icons are intentionally made subtle to prevent visual distraction.

MisterMakerNL, 2023-12-23:
The overlay system has tabs, it's just a little bit complicated to setup correctly.

pierreporte, 2024-01-10:
I had a discussion about this on Discord.

There should be an option to define keyboard shortcut to display a specific tab. Also, it would be good if the context menu has an option for properties, as requested, that would also open the tab.

PaddleStroke, 2024-01-11:
Immo these are 2 separate feature requests.
Properties should be available as a temporary dialog that opens on F1 and rightclick/properties.
Then this tab project is another separate feature.

pierreporte, 2024-01-11:
@PaddleStroke It would make sense to have a separate issue because tabs are more general. Indeed I introduced it with Ondsel Lens as an example of application. What do you mean by a temporary dialog?  Do you want it eventually replaced by a tab, that is making a dialog as a temporary solution?

PaddleStroke, 2024-01-11:
No by temporary I mean a floating dialog that the user open with the right
click menu action. And close when he's done.
As opposed to the dock views which are docked.
What is shown in the first post screenshot basically.

adrianinsaval, 2024-01-17:
I don't see the point on having a temporary window, why not just force the regular panel to be visible when activating the command? Then the user can choose if they want it docked or windowed. I personally hate separate floating windows but it would be great for me to be able to quickly show or hide the property view.

adrianinsaval, 2024-01-17:
yes, we have a perfectly functional panel already. why duplicate it?

adrianinsaval, 2024-01-17:
Ok I see the appeal although I personally would prefer the panel. I might be the outlier here though so don't mind me that much

adrianinsaval, 2024-01-17:
On the subject of the tabs, I think this is a more generalized problem for all panels so I would appreciate if someone could create a dedicated issue for it. The overlay widgets has some tab functionality but they are pretty bad, probably would be a good idea to polish that instead of trying to create something from scratch.

pierreporte, 2024-01-17:
I started a discussion with the design working group to determine how the separate issue should be worded, and if it must be about just the tab feature or also about what kind of tabs there should be.

Zolko-123, 2024-01-17:
I think we should call a session of the consultancy group about the expertise of the UI/UX design experts in the user-panel subcommittee to decide whether the intended workflow of the particular hybrid workflow as proposed is consistent with our gender-neutral policy of climate-change activists decision making process.

Did I get all the bullshit-bingo check-marks ? What did I miss ?

Zolko-123, 2024-01-17:
darn !!! I used "expert" twice, not good

MisterMakerNL, 2024-01-17:
> Well my reason behind was that the PropertyView can be in a lot of places and toggling it can disrupt the UI depending on where it appears. Also it can be a dockwidget or a overlay widget. So overall it felt more messy. It opens in a random place, you may have to resize it depending on where it appears, then finding the close button, the style of which depends on where the view is. Sometime you need to drag it out of somewhere before being able to close it.
> 
> So overall a separate dialog seemed cleaner and behaving as expected. It just opens where you expect, with a size that makes sense and you can close it. Nothing more. And codewise it's not adding much as the PropertyView widget is reused. Also we could use Qt::Popup if we want to make it even easier to dismiss.

Totally agree with this!

Now we maybe need to add some stuff to right click in the future (or somewhere else) where users needed to go to the properties tab for.
In example: 

> I use the properties panel 99% of the time to set the refine state to on. 
> So adding refine to right-click would fix that.  
> Would be nice that I don't have to go too the correct panel, scroll true a list, click on the cell and set it too true. Instead just right-click and boom.

Maybe other users also used it mainly for the same stuff and we can just put in a better accessible UI. 

I really see the properties window as a register-edit function, and in my opinion it shouldn't be necessary to use unless you are doing some freaky stuff.

PaddleStroke, 2024-01-18:
I agree, property view has become a necessity for some user because this or that functionality was missing. If things are at the correct position, property view would be a dev tool and/or very special needs that the UI can't accomodate.

DrD4ffy, 2024-01-18:
For me the property view is not important enough to get permanent screen space, because its mostly not needed in PartDesign. Most things are done with dedicated tasks. But on the other hand its to important to not have quick access to it since addons like Honeycomb, Bevel or EasyReflector but also the built-in SubShapeBinder rely on the property view to be configured. I can drag the property view out of the dock, thats dialog enough. The only missing thing is an easy accessible switch in the interface to show the "property dialog" which then appears where I closed it the last time. When done with editing I can close it again.

closed 2024-01-19 by chennes

maxwxyz, 2024-02-25:
to add context from the complaint session: 
Why does the property panel always fight with the tree? 

- They should be two things.
- Property panel and the tree could be visible at the same time
- Or make the property panel linked again to a keyboard accelerator --> done (Alt + Enter)
- When I want to see properties give me a big window and let me see all the properties.
- Maybe a hot key to unfold properties and hide the model.
- Based on that, tree view should be somewhere alone, transparent on the side, on the screen somewhere, and not get in the way of the rest of the UI.

Maybe the default layout could be changed, currently everything is crammed on the left side, task view on top, then tree view, then property view. @obelisk79