FreeCAD/FreeCAD #11082: [Problem] Broken PartDesign fillet

Anthony-Gaudino, 2023-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to fillet this edge to 1/64" breaks the part:
[image]
[image]

For some reason, trying the same edge on the other side works.
Part: [broken_fillet.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/12950237/broken_fillet.FCStd.zip)

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.8.18
  * Silk 0.1.5
  * Defeaturing 1.2.1
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.66
  * lattice2 1.0.0
  * sheetmetal 0.3.1
  * toSketch 1.0.1
  * Curves 0.6.13
  * Assembly4 0.50.4
  * Assembly3 0.12.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-31:
Re-added 0.21 About info (this is also important to keep because it shows that this is may not be regression (of course there may be the case when the regression could have happened in an earlier version <0.21.x))

pixelwaster, 2024-09-02:
Can confirm on Win10 (0.22.0dev.38555) that the first fillet in PD does what is shown. If you then ^Z and try the other fillet it works. 
If trying the other fillet first, looks to work but it doesn't. The two faces on the side (Face8 and Face9) become a single face, and the fillet has a "hole" that lets the bottom face be selected though it. **fillets smaller than 0.015625" (0.0125") work.

Report window
```
opening file
15:11:31  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: broken_fillet#broken_fillet
15:11:31  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: broken_fillet#Body
15:11:31  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: broken_fillet#BaseFeature

adding fillet
15:12:30  Chamfer: BRep_API: command not done
```

[image]

`OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * CAM
  * Draft
  * Fem
  * Help
  * Idf
  * Import
  * Inspection
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * Plot
  * Points
  * ReverseEngineering
  * Robot
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web
`

luzpaz, 2024-09-02:
Looks OCC related as usual.

ElementW, 2026-08-25:
Chamfer or fillet here still creates incorrect and invalid geo as of faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and either OCC 7.9.3 or 8.1.0.dev1-3d097a0328

[image]

[image]

Specifically this starts happening once the chamfer/fillet goes past the small recess

[image][image]